YosysHQ/apicula #2: tiled_fuzzer.py: ValueError: Unknown type

derekmulcahy, 2019-12-22:
I get a "ValueError: Unknown type at 0x2110d2" when running tiled_fuzzer.py. I have version 1.9.2.02 of the GOWIN IDE installed.

The output for DEVICE=GW1NR-9 is.

````check 19199191
Traceback (most recent call last):
  File "tiled_fuzzer.py", line 237, in <module>
    fse = fuse_h4x.readFse(f)
  File "/home/derek/apicula/fuse_h4x.py", line 15, in readFse
    tiles['header'] = readOneFile(f, ttyp)
  File "/home/derek/apicula/fuse_h4x.py", line 72, in readOneFile
    raise ValueError("Unknown type at {}".format(hex(f.tell())))
ValueError: Unknown type at 0x2110d2
````

DEVICE=GW1N-1 is similar

````check 19199191
Traceback (most recent call last):
  File "tiled_fuzzer.py", line 237, in <module>
    fse = fuse_h4x.readFse(f)
  File "/home/derek/apicula/fuse_h4x.py", line 15, in readFse
    tiles['header'] = readOneFile(f, ttyp)
  File "/home/derek/apicula/fuse_h4x.py", line 72, in readOneFile
    raise ValueError("Unknown type at {}".format(hex(f.tell())))
ValueError: Unknown type at 0x162792
````

pepijndevos, 2019-12-22:
Ai, I'm on 1.9.1.01, so it appears that they added a new type to the `.fse` file. It'll have to be added to the if statement in `readOneFile`.

If you open the `.fse` file in a hex editor, and go to the offset in the specified error message, you can try to look for any patterns. It's _probably_ a `shortval` or `longval`. You could also try adding the type to any of the cases, and if the rest of the file parses correctly, it's probably the correct one.

pepijndevos, 2019-12-22:
Could you amend the error message to print the type, and try adding `0x4e` to parse as a `logicinfo`?

derekmulcahy, 2019-12-22:
Both of the values are 00.

ValueError: Unknown type at 0x2110d2
Contents of GW1NR-9.fse on 1.9.2.02 is

```
002110c0  01 00 00 00 00 00 00 00  00 00 4e 00 00 00 01 00  |..........N.....|
002110d0  00 00 00 00 00 00 00 00  07 00 00 00 1c 00 00 00  |................|
002110e0  3c 00 00 00 07 00 00 00  02 00 00 00 a9 0b 00 00  |<...............|
```

ValueError: Unknown type at 0x162792
Contents of GW1N-1.fse on 1.9.2.02 is

```
00162780: 0100 0000 0000 0000 0000 4E00 0000 0100  ..........N.....
00162790: 0000 0000 0000 0000 0C00 0000 1800 0000  ................
001627a0: 3C00 0000 0700 0000 0200 0000 790B 0000  <...........y...

```

pepijndevos, 2019-12-22:
Ah, but it reads 2 4-bit ints before it errors, so the offset where it read the type is a bit earlier. There is indeed a 4e in there. Pretty sure that if you add that as a logicinfo, it'll work perfectly.

I'm packing for a flight tomorrow, so if you could send a PR, I'd be much obliged. Might be good to add some extra info to the error message while you're at it, for the next unfortunate soul trying it on a future vendor software release.

btw, I'm impressed by how fast you got here. My plan was to silently make the repo public and announce it on Christmas. Shhhht! haha

derekmulcahy, 2019-12-22:
I changed 
     `raise ValueError("Unknown type at {}".format(hex(f.tell())))`
to
      `raise ValueError("Unknown type {} at {}".format(typ,hex(f.tell())))`

and for the two devices I get
```
ValueError: Unknown type 78 at 0x2110d2
ValueError: Unknown type 78 at 0x162792

derekmulcahy, 2019-12-22:
Adding 0x4e to the typ list just produces the following error for both devices.
```
        elif typ in {7, 8, 9, 10, 0xb, 0xc, 0xd, 0xe, 0xf, 0x10,
                     0x27, 0x31, 0x34, 0x37, 0x39, 0x3b, 0x3e,
                     0x3f, 0x41, 0x43, 0x46, 0x48, 0x4a, 0x4c, 0x4e}:
            typn = "logicinfo"
            t = readTable(f, size, 3, 2)
```
```
check 19199191
Traceback (most recent call last):
  File "tiled_fuzzer.py", line 249, in <module>
    pin_names = pindef.get_locs(device, params['package'], True, params['header'])
  File "/home/derek/apicula/pindef.py", line 32, in get_locs
    df = get_package(series, package, special_pins, header)
  File "/home/derek/apicula/pindef.py", line 14, in get_package
    assert fname, "No file found for {}".format(series)
AssertionError: No file found for GW1NR-9
```

derekmulcahy, 2019-12-22:
I knew you were going to pull the trigger on the release at some time.

 I will put together a PR. Maybe it shouldn't die for unknown types. Happy to contribute.

pepijndevos, 2019-12-22:
Perfect!

I don't think there is a way to not die, because different types have different sizes, so you don't know where the next type starts because it's encoded as number of items, not number of bytes.

The fine not found error is a different issue: You still need to download the pinout excel files from Gowin and put them in `~/Documents/gowinsemi/`. Open to better solutions... the same data can probably be extracted from the `.dat` file, but this requires some work. Reading excel files is easier...

derekmulcahy, 2019-12-23:
I downloaded the 1.9.1.01 software, `UG107-1.07E_GW1N-1 Pinout.xlsx` and `UG801-1.5E_GW1NR-9 Pinout.xlsx` files.

I also installed python3.7. `DEVICE=GW1N-1` produces.

```
Traceback (most recent call last):
  File "tiled_fuzzer.py", line 249, in <module>
    pin_names = pindef.get_locs(device, params['package'], True, params['header'])
  File "/home/derek/apicula/pindef.py", line 32, in get_locs
    df = get_package(series, package, special_pins, header)
  File "/home/derek/apicula/pindef.py", line 17, in get_package
    df = df.dropna(subset=[package])
  File "/home/derek/gowin/env/lib/python3.7/site-packages/pandas/core/frame.py", line 4859, in dropna
    raise KeyError(list(np.compress(check, subset)))
KeyError: ['QN48']
```

and `DEVICE=GW1NR-9` produces

```
Traceback (most recent call last):
  File "tiled_fuzzer.py", line 274, in <module>
    for m in zip_longest(*modmap.values(), fillvalue=codegen.Module())]
  File "tiled_fuzzer.py", line 274, in <listcomp>
    for m in zip_longest(*modmap.values(), fillvalue=codegen.Module())]
TypeError: sum() takes no keyword arguments
```

pepijndevos, 2019-12-24:
For the first error, the key is a package column in the "Pin List" tab. If they changed the package name somehow, it should be sufficient to pick pretty much any column. (the more pins the better actually) These are defined in the `params` dict at the top of `tiled_fuzzer.py`.

Apparently the start parameter is new in Python 3.8: https://docs.python.org/3/library/functions.html#sum
Happy to accept a patch that combines the modules in a more compatible way.

I'll announce the repo tomorrow. If you manage to make it work with the latest files and a dated Python, that would be amazing, else I'll for now update the readme to specify the versions needed.

pepijndevos, 2019-12-24:
Currently running `riled_fuzzer.py` from my VPS, and tested all your fixes, but can't reproduce the key error on a fresh download and fresh installation.

Can you confirm the latest commit works for you, or maybe do a bit more debugging on the keyerror to show what's going on? For example, print all the column names in `pindef.py`

derekmulcahy, 2019-12-24:
I had downloaded more pinout files so it was finding the wrong one. It was finding `UG167-1.1E_GW1N-1S Pinout.xlsx` instead of 'UG107-1.07E_GW1N-1 Pinout.xlsx'.

I guess that the -1S variant isn't available in a QN48 package.

pepijndevos, 2019-12-24:
A, good catch. I'd happily accept a PR that updates the code to be more selective in picking the right file. Will close this issue now, as the original problem has been fixed.

closed 2019-12-24 by pepijndevos