YosysHQ/apicula #4: openFPGALoader for JTAG config ?

Martoni, 2019-12-29:
Hi,
Have you tested openFPGALoader for SRAM configuration of bitstream ?
[It's seem's to be compatible ](https://github.com/trabucayre/openFPGALoader) with Gowin FPGA.

pepijndevos, 2019-12-29:
No, I have not. Thanks for pointing it out. I'll test it after the holiday and update the instructions if it works. Also happy to hear the experience of anyone else who tried it.

trabucayre, 2020-01-04:
Hi,
Just to complete Martoni's message: openFPGALoader is now able to write bitstream in flash.

pepijndevos, 2020-01-18:
Cool, I tested it and updated the readme instructions. Hit a small issue, but it generally seems to work.

Have not yet tested it on Tang Nano, so I'll leave the vendor tool in there for now as an alternative.

closed 2020-01-18 by pepijndevos

trabucayre, 2020-01-18:
Hi,
Tang Nano is WIP. I'm able to load bitstream to the SRAM but I've some hang when I try to write in flash.
Maybe I need to push current state with note about limitations.
Thank for the readme update