YosysHQ/apicula #6: tiled_fuzzer.py: ModuleNotFoundError: No module named '_bz2'

Martoni, 2020-03-09:
I think my python installation is wrong but can found why. With python3.7 (virtualenv) I have this exception in fuzzer step : 
```
$ python tiled_fuzzer.py # makes $DEVICE.pickle
Traceback (most recent call last):
  File "tiled_fuzzer.py", line 17, in <module>
    import pindef
  File "/home/fabien/myapp/apicula/pindef.py", line 1, in <module>
    import pandas as pd
  File
"/home/fabien/pyenv/pyenv37/lib/python3.7/site-packages/pandas/__init__.py",
line 54, in <module>
    from pandas.core.api import (
  File
"/home/fabien/pyenv/pyenv37/lib/python3.7/site-packages/pandas/core/api.py",
line 29, in <module>
    from pandas.core.groupby import Grouper, NamedAgg
  File
"/home/fabien/pyenv/pyenv37/lib/python3.7/site-packages/pandas/core/groupby/__init__.py",
line 1, in <module>
    from pandas.core.groupby.generic import DataFrameGroupBy, NamedAgg,
SeriesGroupBy
  File
"/home/fabien/pyenv/pyenv37/lib/python3.7/site-packages/pandas/core/groupby/generic.py",
line 60, in <module>
    from pandas.core.frame import DataFrame
  File
"/home/fabien/pyenv/pyenv37/lib/python3.7/site-packages/pandas/core/frame.py", 
line 104, in <module>
    from pandas.core.generic import NDFrame, _shared_docs
  File
"/home/fabien/pyenv/pyenv37/lib/python3.7/site-packages/pandas/core/generic.py",
line 99, in <module>
    from pandas.io.formats import format as fmt
  File
"/home/fabien/pyenv/pyenv37/lib/python3.7/site-packages/pandas/io/formats/format.py",
line 75, in <module>
    from pandas.io.common import stringify_path
  File
"/home/fabien/pyenv/pyenv37/lib/python3.7/site-packages/pandas/io/common.py",
line 3, in <module>
    import bz2
  File "/usr/local/lib/python3.7/bz2.py", line 19, in <module>
    from _bz2 import BZ2Compressor, BZ2Decompressor
ModuleNotFoundError: No module named '_bz2'
```

I installed the packet libbz2-dev but with the same error.
```
$ sudo apt-get install libbz2-dev
```

pepijndevos, 2020-03-09:
How did you install Python? Is it just from the repos, or compiled from source? I'm assuming from source, given the error.
After you install the bz2 headers, be sure to run the Python configure script again and recompile. The configure script is what detects system libraries and selects the appropriate modules.

Martoni, 2020-03-09:
Yes it was from source. I'll try to relaunch configure this evening.

Martoni, 2020-03-11:
Problem solved, thanks.

closed 2020-03-11 by Martoni

pepijndevos, 2020-03-11:
Glad to hear. I hope you can get a blinky going now.

btw, your question on Twitter prompted me to start looking into Docker. I managed to publish an image to Docker hub, so next step is trying to set up automatic generation of the database https://hub.docker.com/repository/docker/pepijndevos/apicula