YosysHQ/apicula #7: next-pnr python version does not match virtualenv version.

Martoni, 2020-03-11:
I'm using virtualenv version to use python 3.7 : 
```shell
$ source ~/pyenv/pyenv37/bin/activate
$ python --version
Python 3.7.5
```
But nextpnr use python 3.5 to run python script given in argument:
```shell
$  echo "import sys;print(sys.version)" > version.py
$ nextpnr-generic --run version.py 
3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516]
```
Is it a compile option for nextpnr ? Or can I change it with environment vars ?

3.5 is too old for executing simple.py script in fact (f-string problem).
```
 nextpnr-generic --pre-pack simple.py --pre-place simple_timing.py --json blinky.json --post-route bitstream.py --write pnrblinky.json
  File "simple.py", line 18
    with open(f"../{device}.pickle", 'rb') as f:
                                  ^
SyntaxError: invalid syntax
ERROR: Error occurred while executing Python script simple.py
0 warnings, 1 error
``` 

I keep trying, but if you have a clue ;)

Martoni, 2020-03-11:
Solved ;)
But I had to change the CMakeList.txt of nextpnr project : 
```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 487db42..84a79c6 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -103,10 +103,10 @@ if (BUILD_GUI AND NOT BUILD_PYTHON)
     message(FATAL_ERROR "GUI requires Python to build")
 endif()
 
-find_package(PythonInterp 3.5 REQUIRED)
+find_package(PythonInterp 3.7 REQUIRED)
 if (BUILD_PYTHON)
     # TODO: sensible minimum Python version
-    find_package(PythonLibs 3.5 REQUIRED)
+    find_package(PythonLibs 3.7 REQUIRED)
 else()
     add_definitions("-DNO_PYTHON")
 endif()
```

And re-build/re-install nextpnr (don't forget to delete CMakeCache.txt before)

Martoni, 2020-03-11:
But can't import numpy : 
```
$ nextpnr-generic --run version.py  -v --debug
Traceback (most recent call last):
  File "version.py", line 2, in <module>
    import numpy;
ModuleNotFoundError: No module named 'numpy'
ERROR: Error occurred while executing Python script version.py
0 warnings, 1 error
```

Ach, so close to my blinking LEDs ;)

pepijndevos, 2020-03-12:
Nextpnr does not use the virtualenv, so you'll need to install numpy globally.

This definitely needs to be updated in the readme. (Or refactoring the generic script to have less requirements) Would gladly accept a PR.

Martoni, 2020-03-12:
ok my python3 installation is really fucked. In fact that works globally:
```
$ sudo python3 -m pip install numpy
```

Thanks

closed 2020-03-12 by Martoni