YosysHQ/apicula #9: LEDs doesn't blink

Martoni, 2020-03-12:
I followed the tutorial given in main README. The bitstream pack.fs has been generated and downloaded with openFPGALoader but LEDs doesn't blink. All LED are on.

I used official «pinout» from Gowin website maybe it's that ?

I have an error after downloading the bitstream but I read on [this issue](https://github.com/trabucayre/openFPGALoader/issues/14) that is a known bug.

pepijndevos, 2020-03-12:
Can you share the output of the unpack and show steps? Show can be told to output a PNG or SVG rather than show a window. The key thing to look for is if the inputs are fully connected to the outputs.

If there is nothing to render, this is a very bad sign. Remove the purge argument and try again. This will produce a HUGE graph. You can look at the source code and search for the IOB locations. As mentioned, I have a hunch that your IOB did not get created correctly.

You can also pass a selector to show, to only show the IOB, or you can of course just look at unpack.v and search around.

Martoni, 2020-03-12:
Solved by trabucayre [PR](https://github.com/pepijndevos/apicula/pull/10) !

closed 2020-03-12 by Martoni

pepijndevos, 2020-03-12:
Cool, so you have a working blinky! Yay!

Martoni, 2020-03-12:
https://www.youtube.com/watch?v=yc2yBYlCCI4&feature=youtu.be

\o/