YosysHQ/apicula #12: bash simple.sh nanolcd/*.v # Tang Nano

paulossant, 2020-06-07:
When following the quick start up instructions for the sipeed tang nano, when running nr$ bash simple.sh blinkygw1n1.v # Tang Nano
It returns (at the end):
Info: Routing..
Info: Setting up routing queue.
ERROR: Found two arcs with same sink wire R11C10_A0: ctr[23] (3) vs ctr[24] (2)
ERROR: Routing design failed.
0 warnings, 2 errors

Note I tried the previous steps with several available version of gowin IDE

best regards,
Paulo

pepijndevos, 2020-07-01:
Eh, what are your environment variables set to? Which versions (Python, Gowin IDE, the data files, etc.) are you on?

pepijndevos, 2020-08-03:
I assume this was fixed with #13 and caused by changes in nextpnr

closed 2020-08-03 by pepijndevos