YosysHQ/apicula #18: Update documentation

pepijndevos, 2020-09-18:
There is no documentation on what is in the timing file that was recently decoded.

The readme is also out of date about some recent work.

I also need to add the IRC channel, and eventually Patreon link.

Anything else?

nalzok, 2020-09-20:
A quick demo of how to *use* this project would be nice. I'm on macOS so it's impossible to install the GoWin IDE as the current README instructs, but ideally this project should enables a cross-platform toolchain for GoWin FPGAs, just like Project IceStorm does for ICE40?

pepijndevos, 2020-09-20:
Right, we are now generating the bitstream in a Github action, and building a Docker image to do so. So you could either download the generated file, which would let you use the toolchain, but not really contribute. Or you can install the Docker image and generate your own chipdb and contribute changes. Thanks for the reminder, I will add this to the updated readme.

nalzok, 2020-09-20:
You mean I can contribute by just running a Docker image on my MacBook? That's cool. Will contribute some computational power when the instructions on how to contribute with Docker is ready :P

pepijndevos, 2020-10-01:
Okay, I updated stuff a bit. No doubt there will be more to update in the future.

closed 2020-10-01 by pepijndevos