YosysHQ/apicula #19: FPGA guide

peepo, 2020-10-01:
Great to find this project, 
especially as you are maintaining regularly!!

Have been using IceStorm,
thinking about TangNano.

Do you know of a good guide to understanding FPGA (de)construction?
not verilog, but the mechanics, ie like icestorm;

Seems you might be the person to write one?
as you already understand something of this...
cf.  http://pepijndevos.nl/   Contents   page please

Comparison of UpDuino/TangNano might help better understanding?

did not find email details
apologies jay@peepo.com

pepijndevos, 2020-10-02:
What information are you looking for? The resources section of the readme has a link to my internship report and some other documentation where I try to explain how stuff works. I've also done livestreams and recently Jupyter notebooks to give some insight to my process. So I'm trying very hard to write the kind of information you are looking for, but it's hard to know what people want to know. So if you've read all the things I already wrote and are missing something more specific, I'd be happy to clarify.

... no sure what you mean with the cf.

pepijndevos, 2020-12-29:
Closing this because I've not heard anything and it's not really a bug.

closed 2020-12-29 by pepijndevos