YosysHQ/apicula #31: Tang Nano Problem

TheFlav, 2021-06-22:
FYI
On my Tang Nano, when I did all of this.

        cd examples
        yosys -p "synth_gowin -json blinky.json" blinky.v
        nextpnr-gowin --json blinky.json \
                      --write pnrblinky.json \
                      --device GW1NR-UV9QN881C6/I5 \
                      --cst tec0117.cst
        gowin_pack -d GW1N-9 -o pack.fs pnrblinky.json
        # gowin_unpack -d GW1N-9 -o unpack.v pack.fs
        # yosys -p "read_verilog -lib +/gowin/cells_sim.v; clean -purge; show" unpack.v
        openFPGALoader -b tangnano pack.fs # Tang Nano

I could not get it to finish loading.  I tried a bunch of stuff.  I even replaced the firmware of the CH552 chip.

Anyway, after a lot of flailing and reading, I tried some other .fs (from https://github.com/racerxdl/tangnano-yosys-hello)
and that worked fine.  

After that, in apicula/examples, I did a full make.  Then, this one worked
        openFPGALoader -b tangnano blinky-tangnano.fs

So, maybe "pack.fs" isn't what should be on the main README?

pepijndevos, 2021-06-23:
You are passing the wrong device for all the commands. You're not the first, so I GREATLY welcome a PR to the readme that suggests wording that would have made this clear to you.

Or maybe people just skim the readme and we should copy-paste the commands for every supported board.
Or put placeholder commands `<DEVICE>` so that the user is forced to think about the parameters.

Every reference to a device family `GW1N-9`, a specific device `GW1NR-UV9QN881C6/I5` and a specific board `tec0117.cst`, `tangnano` needs to match. Note that you can pass `GW1NR-UV9QN881C6/I5` to `gowin_pack` as well for uniformity.

trabucayre, 2021-06-23:
ASAP I add check to don't flash a bitstream when it's fs IDCode didn't match FPGA IDCode.

TheFlav, 2021-06-23:
As a bit of background, I was trying to understand the toolchain.  I just received the Tang Nano, and I wanted to try to use it on my Mac without needing the official tools and Windows.

The commands I did
    cd examples
    yosys -p "synth_gowin -json blinky.json" blinky.v
    nextpnr-gowin --json blinky.json \
                  --write pnrblinky.json \
                  --device GW1NR-UV9QN881C6/I5 \
                  --cst tec0117.cst
    gowin_pack -d GW1N-9 -o pack.fs pnrblinky.json
    openFPGALoader -b tangnano pack.fs # Tang Nano
were directly copy/paste from the readme.md.  I didn't even consider that I needed to edit the device until you just mentioned it here.  I now see that the readme mentions the supported boards and their devices, but I would not have realized that it was the problem here at all.  I think maybe the readme.md could be updated to specifically tell the reader that they need to add in the <DEVICE> based on their board.

As trabucayre mentioned, adding that check during the flash process would be very helpful, but that'd be in openFPGALoader, right?

trabucayre, 2021-06-24:
I've updated openFPGALoader to check coherence. Now this type of issue will not happens anymore.

pepijndevos, 2021-06-24:
Actually #30 aims to change the readme in pretty much this way, but wasn't merged because it states in very uncertain terms things that should be unambiguous. Does that look good to you? Feel free to chime in with improvements. Maybe I should just merge it and fix it up myself.

TheFlav, 2021-06-24:
I like this addition, "The example below is for the Trenz TEC0117. For other devices, use the device IDs listed above, and replace `tec0117` with `runber` or `tangnano` accordingly."

I also see that he changed the 3 openFPGALoader commands to just one, BUT he should have added a better comment there.

One thing that tripped me up with the readme was that it says "compile a blinky" and then it shows the steps to do that.  It was obvious to me that there were 3 options for the openFPGALoader command, and that made me choose one of them (for my board).  BUT there weren't options for the previous commands.

So, I think maybe something like this is better.

The example below is for the Trenz TEC0117. For other devices, use the device IDs listed above, and replace `tec0117` with `runber` or `tangnano` accordingly.
        cd examples
        make make blinky-tec0117-prog # or blinky-runber-prog or blinky-tangnano-prog

or
        cd examples
        yosys -p "synth_gowin -json blinky.json" blinky.v
        nextpnr-gowin --json blinky.json \
                      --write pnrblinky.json \
                      --device GW1NR-UV9QN881C6/I5 \  #set this to your device like GW1NR-UV9QN881C6/I5, GW1N-LV1QN48C6/I5, GW1N-UV4LQ144C6/I5, etc
                      --cst tec0117.cst  #choose the cst file for your board
        gowin_pack -d GW1N-9 -o pack.fs pnrblinky.json # You may need to change GW1N-9, runber uses GW1N-4 and tangnano uses GW1N-1
        # gowin_unpack -d GW1N-9 -o unpack.v pack.fs
        # yosys -p "read_verilog -lib +/gowin/cells_sim.v; clean -purge; show" unpack.v
        openFPGALoader -b tec0117 pack.fs  #select tec0117, runber, or tangnano

pepijndevos, 2021-07-05:
Ok I updated the readme. I hope this makes it clearer.

closed 2021-07-05 by pepijndevos