YosysHQ/apicula #32: Failed to run nextpnr-gowin for blinky.v example

leeooox, 2021-06-28:
Hello,

```shell
yosys -p "synth_gowin; write_json blinky.json; show -format svg  -prefix aaa;" -f verilog -q blinky.v
nextpnr-gowin --json blinky.json --write pnrblinky.json --device GW1NR-UV9QN881C6/I5 --cst tec0117.cst
```

Error information of nextpnr-gowin  command, is there anything wrong. 

Warning: Invalid constraint: 
Warning: Invalid constraint: IO_LOC "clk" 35;
Warning: Invalid constraint: IO_LOC "rst" 77;
Warning: Invalid constraint: IO_LOC "led[0]" 86;
Warning: Invalid constraint: IO_LOC "led[1]" 85;
Warning: Invalid constraint: IO_LOC "led[2]" 84;
Warning: Invalid constraint: IO_LOC "led[3]" 83;
Warning: Invalid constraint: IO_LOC "led[4]" 82;
Warning: Invalid constraint: IO_LOC "led[5]" 81;
Warning: Invalid constraint: IO_LOC "led[6]" 80;

Info: Packing constants..
Info: Packing IOs..
Info: Packing LUT-FFs..
Info: Packing non-LUT FFs..
Info: Checksum: 0xe0932f19

Info: Annotating ports with timing budgets for target frequency 12.00 MHz
Info: Checksum: 0xe0932f19

Info: Device utilisation:
Info: 	               SLICE:    28/ 8640     0%
Info: 	                 IOB:     9/  274     3%

Info: Placed 1 cells based on constraints.
ERROR: Unable to place cell '$auto$alumacc.cc:474:replace_alu$4.slice[9].alu', no BELs remaining to implement cell type 'ALU'
10 warnings, 1 error

PS: the example in readme is promote errors with latest yosys repo, so I changed  **-json** to **write_json**
```shell
yosys -p "synth_gowin -json blinky.json" blinky.v
```
[blinky.json.txt](https://github.com/YosysHQ/apicula/files/6727365/blinky.json.txt)

pepijndevos, 2021-06-30:
The `-json` flag disables features not yet supported by nextpnr-gowin. It also passes `-nobram -nolutram -nowidelut -noalu`. This is why you're seeing the ALU error. Not sure where the constraint warnings are coming from.

NiekIlmer, 2021-12-17:
This ticket is older, but I was having a similar issue with -json not being recognized. What fixed that for me was installing Yosys from source instead of the lazy route via apt-get. Apt installed yosys/focal 0.9-1build2 which was built before you added the -json flag. I'm expecting these issues may be caused by a similar issue.

closed 2022-01-24 by pepijndevos

leeooox, 2022-02-10:
The root cause is the yosys with apt-get is too old version. git clone the source code and compile it to latest version fixed the problem.