YosysHQ/apicula #49: Blinky example is broken

newAM, 2021-09-18:
The instructions to run the blinky example appear to be broken.

After I run `yosys -p "synth_gowin -json blinky.json" blinky.v` I get this error:
```
 Yosys 0.9+4276 (git sha1 75a4cdfc8a, gcc 10.3.0 -fPIC -Os)

-- Parsing `blinky.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend: blinky.v
Parsing Verilog input from `blinky.v' to AST representation.
blinky.v:3: ERROR: Unimplemented compiler directive or undefined macro `LEDS_NR.
```

This appears to be broken due to this commit: https://github.com/YosysHQ/apicula/commit/ecfdeb9775425eb3f03c60231290d1da5530a8f2

Normally I would offer a PR to fix this but I am still learning verilog/FPGAs, and I do not know how this should be corrected.

Thanks for this project!

yrabbit, 2021-09-18:
Not that it's broken, you just need to specify how many LEDs your board has. For example for my TangNano with three LEDs:

`yosys -D LEDS_NR=3 -p "synth_gowin -json blinky.json" blinky.v`

newAM, 2021-09-18:
Should the README be updated then?  I think that applies to the Trenz TEC0117 example as well.

yrabbit, 2021-09-18:
You're right, I've just never done it that way --- make (or gmake in my case with #DragonFlyBSD) takes care of setting the right number of LEDs by itself.
Well, yeah, the instructions need to be corrected.

yrabbit, 2021-09-19:
considered in PR#50, it would take at least three hours to merge :)

https://github.com/YosysHQ/apicula/pull/50

pepijndevos, 2021-09-19:
OK merged. Thanks for bringing this up. I'm kind of tempted to just update the instructions to run make, but that would be less helpful for other projects.

closed 2021-09-19 by pepijndevos

newAM, 2021-09-19:
Thank you for fixing this so quickly!