YosysHQ/apicula #56: How to configure pin multiplexing on Tang Nano

Eriosies, 2021-11-03:
Is it possible to set dual-purpose pins using these tools?
RECONFIG_N and DONE pins need to be set to dual purpose to use UART to CH552 on the Tang Nano.

Configuring dual-purpose pins/pin multiplexing in the Gowin IDE is described [here](https://www.manualslib.com/manual/2074226/Gowin-Gw2a-Series.html?page=10).

pepijndevos, 2021-11-04:
It's technically possible, there is just no UI for it. They are set in hardware mode like so:

https://github.com/YosysHQ/apicula/blob/e46cc07cabed835642faeec5dd6d4bb0566c24cc/apycula/tiled_fuzzer.py#L678-L684

**Note that it's very easy to brick your FPGA by using important pins.**

So basically currently you'd need to write some Python code to manually enable these bits. I'd happily accept a PR that adds a command line interface to `gowin_pack` to set these pin modes.

pepijndevos, 2021-11-04:
Heh, fixed as I was writing this by @yrabbit in  #58

closed 2021-11-04 by pepijndevos