YosysHQ/apicula #59: nextpnr fails to execute JSON backend with v0.0.1a12

newAM, 2021-11-07:
After updating from v0.0.1a11 to v0.0.1a12 nextpnr-gowin fails to execute the JSON backend.

Args:

```text
yosys -p "synth_gowin -json synth.json" top.v
nextpnr-gowin --json synth.json --write design.json --device GW1N-LV1QN48C6/I5 --cst tangnano.cst
```

Error:

```text
4.29. Executing JSON backend.

Warnings: 770 unique messages, 770 total
End of script. Logfile hash: 4236f24c6b, CPU: user 1.44s system 0.02s, MEM: 55.97 MB peak
Yosys 0.10+1 (git sha1 7a7df9a3b4, gcc 10.3.0 -fPIC -Os)
Time spent: 27% 1x abc (0 sec), 22% 3x techmap (0 sec), ...
ERROR: Unsuported speed grade 'C6/I5'.
0 warnings, 1 error
```

This fails for even a simple LED blinky design, `top.v`:

```verilog
(* top *)
module TOP
(
    input rst,
    input clk,

    output LCD_CLK,
    output LCD_HYNC,
    output LCD_SYNC,
    output LCD_DEN,
    output [4:0] LCD_R,
    output [5:0] LCD_G,
    output [4:0] LCD_B,

    output [7:5] led
);

    wire CLK_SYS;    
    wire CLK_PIX;
    wire LED_R;
    wire LED_G;
    wire LED_B;

    assign CLK_SYS = clk;
    assign CLK_PIX = clk;
    assign led[5] = LED_R;
    assign led[6] = LED_G;
    assign led[7] = LED_B;

    // RGB LED TEST
    reg [31:0]cnt;
    reg [1:0]rgb_data;
    always @(  posedge CLK_SYS or negedge rst  )
    begin
        if (!rst) begin
            cnt <= 32'd0;
            rgb_data <= 2'b00;
        end
        else if (cnt == 12000000) begin
            cnt <= 4'b0;
            rgb_data <= rgb_data + 1'b1;
        end
        else
        cnt <= cnt + 1'b1;
    end
    assign LED_R = ~(rgb_data == 2'b01);
    assign LED_G = ~(rgb_data == 2'b10);
    assign LED_B = ~(rgb_data == 2'b11);

endmodule
```

pepijndevos, 2021-11-07:
Sorry about that 0.0.1a12 requires this branch of nextpnr to work https://github.com/YosysHQ/nextpnr/pull/859

newAM, 2021-11-07:
Works now, thanks! :tada:

closed 2021-11-07 by newAM

pepijndevos, 2021-11-07:
I see you are maintaining a nixpkg for Apycula? Amazing.

newAM, 2021-11-07:
I find that it is a good way to manage the cross section of (yosys, nextpnr, apycula) versions.

Thanks for making this easy to package! :smile: