YosysHQ/apicula #64: The bitstream header corruption?

yrabbit, 2021-11-29:
I leave it here so I don't forget it.

Sometimes the generated image cannot be flashed with openFPGALoader --- it complains about missing IDCODE.

I haven't looked into it yet, but attached two versions of both the original .JSON and the resulting .FS

```
rabbit@fly ~/src/apicula/examples% diff -u ~/tmp/bad/shift-tangnano-bad.fs ~/tmp/good/shift-tangnano-good.fs|head
--- /home/rabbit/tmp/bad/shift-tangnano-bad.fs  2021-11-29 10:51:19.089890000 +1000
+++ /home/rabbit/tmp/good/shift-tangnano-good.fs        2021-11-29 10:49:28.988690000 +1000
@@ -1,4 +1,4 @@
-1011101111100001
+1110100010101110
 1111111111111111
```
[image]
[good.tar.gz](https://github.com/YosysHQ/apicula/files/7614829/good.tar.gz)
[bad.tar.gz](https://github.com/YosysHQ/apicula/files/7614831/bad.tar.gz)

pepijndevos, 2021-11-29:
The first line of the fs is only the checksum as far as I can tell?
https://github.com/YosysHQ/apicula/blob/master/doc/commandstructure.md
https://github.com/YosysHQ/apicula/blob/edb70d5286166e14536f6a5f2604dba19e1f8a72/apycula/gowin_pack.py#L227

pepijndevos, 2021-11-29:
In the bad bitstream the first byte and key are
```
In [1]: bin(187)
Out[1]: '0b10111011'

In [2]: bin(59)
Out[2]: '0b111011'

In [3]: hex(59)
Out[3]: '0x3b'
```
so we enter into 
https://github.com/trabucayre/openFPGALoader/blob/7ceaab012650dba42056568afdb65442fe4c8b40/src/fsparser.cpp#L105

pepijndevos, 2021-11-29:
In the good one they are
```
In [4]: bin(232)
Out[4]: '0b11101000'

In [5]: hex(104)
Out[5]: '0x68'
```
and we skip right over the whole case statement.

So it seems that you got really unlucky with the checksum matching a header command, and this seems like an openFPGALoader bug: it should probably just ignore the first line?

/cc @trabucayre

commit 2db2c6d1 (2021-11-29): first line is no longer a checksum, fixes  #64
closed 2021-11-29 by pepijndevos