YosysHQ/apicula #79: Tang Nano 1K example

chiplet, 2022-01-27:
I just bought a [Tang Nano 1K](https://www.aliexpress.com/item/1005002551785169.html?spm=a2g0o.store_pc_groupList.8148356.2.78981d42dwD89o) board, which to my surprise, was not the same thing as Tang nano. The Tang Nano 1K board has a [GW1NZ-1](https://www.gowinsemi.com/en/product/detail/46/) chip instead of GW1N-1 so naturally the examples don't work on the new board. Is the GW1NZ-1 chip already supported by project apicula and nextpnr-apicula so that the example designs could be ported to this board as well?

yrabbit, 2022-01-27:
If you do manage to load the image into it, it may help to specify the correct pins in tangnano.cst (probably they are different from TangNano)

pepijndevos, 2022-01-28:
Yea it's the same situation as 9C. So this is a good example how to add it https://github.com/YosysHQ/apicula/pull/77

chiplet, 2022-01-28:
This seems like a good way to get to know the tooling a bit better so I'll try to add support for GW1NZ-1 using the mentioned PR as a reference.

pepijndevos, 2022-01-30:
I'd put them in shortval and see if it parses. Since 0x50 is there. Else you'll have open the thing in a hex editor at the error index and look for patterns. Usually it's pretty easy to tell the length of the fields.

chiplet, 2022-02-15:
There's now a working blinky example in #81

closed 2022-03-23 by chiplet