YosysHQ/apicula #83: Is there any reason DFF do not support CE at the moment?

leeooox, 2022-02-17:
Hello,

It seems that the DFF primers do not support clock enable(CE), as of the code.

```python

dffmap = {
    "DFF": None,
    "DFFN": None,
    "DFFS": "SET",
    "DFFR": "RESET",
    "DFFP": "PRESET",
    "DFFC": "CLEAR",
    "DFFNS": "SET",
    "DFFNR": "RESET",
    "DFFNP": "PRESET",
    "DFFNC": "CLEAR",
}

```
https://github.com/YosysHQ/apicula/blob/4e8e72098349aa20c365d4940d13b2995843d0c4/apycula/tiled_fuzzer.py#L143

The vendor datasheet documented that these DFF with E extension are supported by hardware.

- DFFE
- DFFSE
- DFFRE
- DFFPE
- DFFCE
- DFFNE
- DFFNSE
- DFFNRE
- DFFNPE
- DFFNCE

So it there any reason CE is not supported?

PS: latest Yosys seems also support CE.

https://github.com/YosysHQ/yosys/blob/15860000487e2d6748843888b78289f95f3ea46b/techlibs/gowin/cells_map.v#L17

yrabbit, 2022-02-17:
They are supported. It's just that all DFFs have a CE input and variants without CE are obtained by routing this input to VCC. You are looking at the table for the fuzzer, which defines the bits to encode the DFFs themselves, not the routing bits.

leeooox, 2022-02-17:
I guess you are talking about these 3 nodes "CE0", 131: "CE1", 132: "CE2", it make sense, thanks.

https://github.com/YosysHQ/apicula/blob/4e8e72098349aa20c365d4940d13b2995843d0c4/apycula/wirenames.py#L10

closed 2022-02-17 by leeooox