YosysHQ/apicula #99: gowin_pack for GW1N-LV1QN48C6/I5 not working

catkira, 2022-04-02:
When I run the example with device GW1N-LV1QN48C6/I5 I get this error:
'''
 root@DESKTOP-NSB1HR8:/mnt/d/git/apicula/examples# gowin_pack -d $DEVICE -o pack.fs pnrblinky.json
Traceback (most recent call last):
  File "/root/.venv/python3.10/bin/gowin_pack", line 8, in <module>
    sys.exit(main())
  File "/root/.venv/python3.10/lib/python3.10/site-packages/apycula/gowin_pack.py", line 366, in main
    with importlib.resources.open_binary("apycula", f"{device}.pickle") as f:
  File "/usr/lib/python3.10/importlib/resources.py", line 46, in open_binary
    return reader.open_resource(resource)
  File "/usr/lib/python3.10/importlib/abc.py", line 433, in open_resource
    return self.files().joinpath(resource).open('rb')
  File "/usr/lib/python3.10/pathlib.py", line 1117, in open
    return self._accessor.open(self, mode, buffering, encoding, errors,
FileNotFoundError: [Errno 2] No such file or directory: '/root/.venv/python3.10/lib/python3.10/site-packages/apycula/GW1NNone-1.pickle'
'''

yrabbit, 2022-04-02:
Try the master branch

catkira, 2022-04-02:
Ah i found the problem in gowin_pack.py, the version on pip does not have 
mods = m.group(1)
and not 
mods = m.group(1) or ""

I thought pip install downloads the latest version from git

catkira, 2022-04-02:
ah the last 2 commits were not in the release

closed 2022-04-02 by catkira

RGD2, 2022-06-02:
This is fixed after doing a `pip install -U apycula`, with version 0.3.1, but other things like yowasp start complaining because they've specified version =0.3 not >=0.3. 
Oh well, all seems to work. Bit worrying that it's still 'broken' at the moment, if getting apicula installed via yowasp-nextpnr-gowin, so I'll go make a ticket there, if there isn't already.

RGD2, 2022-06-04:
Had a chat with them, they're going to keep using exact version match deps, with only manually selected publication of their nightly builds. 

But I think I prompted them to at least consider auto publication of nightly releases where you guys to do a 0.0.+1 version bump, specifically for quick little fixes like this that shouldn't break compatibility. 

Anyway, they manually pushed an update for 0.3.1, so it's all good for now.