YosysHQ/apicula #105: Tang Nano 9k support?

chenzhuoyu, 2022-05-25:
Is there any plans to support the Tang Nano 9k board?

chenzhuoyu, 2022-05-25:
@pepijndevos That's nice. But it does not listed in the README.md, so it probably needs to be updated?

pepijndevos, 2022-05-25:
Correct. Hmmm if you have the thing sitting on your desk feel free to add it to the list with the correct part number, else I'll do it myself later.

chenzhuoyu, 2022-05-25:
@pepijndevos I don't own a Tang Nano 9k at this moment, but I'm considering buying one. Once it arrives I'll give it a try.

fanoush, 2022-05-27:
Hi, just received this board today :-) This is what openFPGAloader shows with tang nano 9k
```
openFPGALoader.exe --detect
←[94mJtag frequency : requested 6.00MHz    -> real 6.00MHz   ←[0m
index 0:
        idcode 0x100481b
        manufacturer Gowin
        family GW1N
        model  GW1N(R)-9C
        irlength 8
```
and here is the chip
[image]
hope it helps

pepijndevos, 2022-05-27:
oh yea forgot about this, thanks! saves me a trip upstairs to fetch my board, added : https://github.com/YosysHQ/apicula/commit/4b62db6848037dffc3569cc1fc5a47c463e54a94

pepijndevos, 2022-05-27:
Needs an extra family flag though, because the C is not in the device number: https://github.com/YosysHQ/apicula/blob/4b62db6848037dffc3569cc1fc5a47c463e54a94/examples/Makefile#L79

Should find a way to work that into the readme

closed 2022-05-27 by pepijndevos