YosysHQ/apicula #108: Any tools/examples for the PLL primitive?

chenzhuoyu, 2022-05-29:
For `ice40` chips there is [icestorm/icepll](https://github.com/YosysHQ/icestorm/tree/master/icepll), for `ecp5` chips there is [trellis/ecppll](https://github.com/YosysHQ/prjtrellis/blob/master/libtrellis/tools/ecppll.cpp).

I wonder is there any similar tools/examples for `gw1n` chips?

pepijndevos, 2022-05-29:
No, unfortunately PLL is not yet supported. In case you'd like to add support for it, hop into the Matrix channel.

chenzhuoyu, 2022-05-29:
Is there any plans? My skill level is far from implementing it, so I might just wait...

yrabbit, 2022-05-29:
Yes, implementation of this primitive is in the plans. 
And if one part, namely the conversion of the Verilog text into bits for FPGA firmware, requires some perverse knowledge, the second part, namely the generation of the text primitive based on the specified characteristics of the frequency and other things, I think enough knowledge of the equations of the first order and the basics of Python.

Come to the Matrix chat room.

closed 2022-08-15 by pepijndevos