YosysHQ/apicula #117: tangnano9k: FPGA device: mismatch between target's idcode and fs idcode

hvegh, 2022-06-26:
Just recieved my tangnano9k and tried to upload simple led demos.

Below script runs 100% ok except for the bitstream upload... It seem the bitstream generated is for different device...

```
yosys -D LEDS_NR=8 -p "read_verilog blinky.v; synth_gowin -json blinky.json"
DEVICE='GW1NR-LV9QN88PC6/I5'  # change to your device
BOARD='tangnano9k' # change to your board
nextpnr-gowin   --json blinky.json \
                --write pnrblinky.json \
                --family GW1N-9C \
                --device $DEVICE \
                --cst $BOARD.cst
gowin_pack -d $DEVICE -o pack.fs pnrblinky.json # chango to your device
# gowin_unpack -d $DEVICE -o unpack.v pack.fs
# yosys -p "read_verilog -lib +/gowin/cells_sim.v; clean -purge; show" unpack.v
openFPGALoader -v -b $BOARD pack.fs
```

results in:

```
Yosys 0.18+29 (git sha1 b2408df31, clang 13.0.1 -fPIC -Os)
Time spent: 38% 10x read_verilog (0 sec), 12% 13x opt_clean (0 sec), ...
+ DEVICE=GW1NR-LV9QN88PC6/I5
+ BOARD=tangnano9k
+ nextpnr-gowin --json blinky.json --write pnrblinky.json --family GW1N-9C --device GW1NR-LV9QN88PC6/I5 --cst tangnano9k.cst
...
Info: Program finished normally.
+ gowin_pack -d GW1NR-LV9QN88PC6/I5 -o pack.fs pnrblinky.json
+ openFPGALoader -v -b tangnano9k pack.fs
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
found 1 devices
index 0:
	idcode 0x100481b
	manufacturer Gowin
	family GW1N
	model  GW1N(R)-9C
	irlength 8
File type : fs
Parse file Parse pack.fs: 
checksum 0x3625
Done
DONE
bitstream header infos
CRCCheck: ON
Compress: OFF
ConfDataLength: 712
ProgramDoneBypass: OFF
SPIAddr: 00fff000
SecurityBit: ON
idcode: 1100581b
loading_rate: 0

```

hvegh, 2022-06-26:
Checked the openFPGAloader sources:
```
fsparser.cpp:		case 0x1100581b: /* GW1N-9    */
fsparser.cpp:		case 0x1100481b: /* GW1N-9C   */

```

Is this due to nextpnr-gowin missing to parse the C in family "GW1N-9C" ?

yrabbit, 2022-06-26:
Can you repeat this with the master branch of nextpnr? Because I'm only two commits behind the master and I don't see this problem.
[image]

pepijndevos, 2022-06-26:
You need to run `gowin_pack` with the correct family as well, because again the C is only in the date code. Have a look at the examples makefile for the correct command.

hvegh, 2022-06-26:
Yes, thank you! 

Total missed checking parameters to gowin_pack, this works fine now! 
```

yosys -D LEDS_NR=8 -p "read_verilog blinky.v; synth_gowin -json blinky.json"

DEVICE='GW1NR-LV9QN88PC6/I5'  # change to your device
FAMILY='GW1N-9C'   # change to your family
BOARD='tangnano9k' # change to your board

nextpnr-gowin   --json blinky.json \
                --write pnrblinky.json \
                --family $FAMILY \
                --device $DEVICE \
                --cst $BOARD.cst

gowin_pack -d $FAMILY -o pack.fs pnrblinky.json

# gowin_unpack -d $FAMILY -o unpack.v pack.fs
# yosys -p "read_verilog -lib +/gowin/cells_sim.v; clean -purge; show" unpack.v

openFPGALoader -v -b $BOARD pack.fs

```

pepijndevos, 2022-06-26:
Open to a PR that makes it more obvious in the readme how to do this. Tricky thing is for now tangnan9k is the exception here.

closed 2022-06-26 by pepijndevos