YosysHQ/apicula #123: tang nano 1k program is volatile?

Joshinken, 2022-08-13:
for some reason the code uploaded to the device will disappear the moment that the device is powered down. is that supposed to happen? how do i make it not do that?

Joshinken, 2022-08-13:
turns out its openfpgaloader, if you want the program to write to the internal flash of the tang nano you need to use -f. id write something like this in the readme, but -f is only supposed to be used on some boards, i took a gamble trying it on the tang nano which was not listed by openfpga as a board that supports this, and i dont have any other boards to test it with

Joshinken, 2022-08-13:
eh. maybe i should keep it open in case someone else has the same problem and feels lucky

pepijndevos, 2022-08-15:
Well, it's not really a bug so I'll close it, but if you want to make a PR to the readme to clarify that would be possible.

closed 2022-08-15 by pepijndevos

Joshinken, 2022-08-15:
> Well, it's not really a bug so I'll close it

fair enough, i thought it would be ok since the other 4 open issues arent really issues either. I’ll look into what exactly the problem with openfpgaloaders -f is and write a PR, probably gonna add it to the openfpgaloader wiki too.