YosysHQ/apicula #127: Tang Primer 20k support

rafaelmartins, 2022-09-07:
Hello,

there are any plans to support the new Tang Primer 20k board (GW2A-LV18PG256C8/I7)?

As far as I can see, we don't have any GW2A support yet. Could we discuss the steps required to get this included? I have the hardware and would be willing to help.

Thanks.

pepijndevos, 2022-09-07:
Awesome! 

So in the basis you could head over to the closed pull requests and find some that added other devices, start along that path and see what breaks. And then do the same on the nextpnr side.

Hopefully they are kinda just bigger versions of Littlebee, but no doubt there will be some issues along the way.

One problem that should be dealt with *eventually* is that the nextpnr target constructs the chipdb at runtime rather than memory-mapping a big struct as some other targets do. This means that the bigger the device, the longer the loading time. For 9k this is already a few second, and I imagine for 20k and 55k it will be kinda extremely slow. But rewriting the nextpnr target is a much bigger project than just adding a new devices.

pepijndevos, 2022-09-07:
Also, I'd recommend joining the Matrix room to hang out and ask questions.

rafaelmartins, 2022-09-10:
Ok, thank you for the insights. I'll start looking at it this weekend and direct questions to the matrix chat :)

takanuva, 2022-10-16:
Any progress on this?

pepijndevos, 2022-10-17:
Yea there has not been much progress on this front to my knowledge. Buying the FPGA myself is on my todolist, but if you want to jump in and help, the same applies: look at other PRs that added a device, join Matrix/IRC

rafaelmartins, 2022-10-17:
I started working on this, and already understand what needs to be done / implemented a few changes, but I'm taking my time to learn more about how the toolchain / fuzzers / official gowin software works, as these implementation details are kinda new for me. I'll publish my first changes soon, but any help here from experienced people is more than welcome.

pepijndevos, 2022-10-19:
Amazing news. If you have any questions, don't hesitate to stop by on Matrix/IRC

blark, 2023-03-24:
With the upcoming Tang Nano 20k, and the Primer 20k it will be _really_ great to have open source toolchain support. Happy to test this out if that helps.

mdelcimar, 2023-06-16:
Hello;
  I recently acquired tangprimer20k but this is one of my first experiences with fpga, I've been studying it for about 3 months and I know how to describe hardware for at least 3 years but I've never ventured into the area, I know that language here and a limitation for us to communicate.
   Well, I'm a system analyst, I've been programming for a long time to tell the truth and my training, but I've moved on to the world of linux admins but I haven't lost the habit of always programming argo for any problem I have, I'd also like to contribute to the project from the tang primer 20k the one I have because I only have it to test, if you can help me with the forwarding, I have no idea how you analyze the closed system to try to make an open one, I have my ways of trying to do reverse engineering but it would be good some tips.

AT Delcimar Martins

yrabbit, 2023-06-16:
I compile the test pieces with the vendor IDE and look at the output binary images in the Python interpreter.
But I'm an amateur, but you can see how the pros do it  -     https://www.youtube.com/watch?v=Vt7FyOXfkZA&list=PLIYslVBAlKZad3tjr5Y4gqBV3QKQ5_tPw&index=1   - and read -   https://github.com/YosysHQ/apicula/blob/master/clock_experiments.ipynb

pepijndevos, 2023-06-16:
This is great to hear! I would suggest that you join the matrix/irc channel linked in the readme.

A good place to start is to look at previous PRs that added support for a new FPGA model. There are some places where values need to be added from documentation or experiments, and then probably a ton of bugfixes.

yrabbit, 2023-06-17:
It's very likely that we will get the support of tangprimer20k today !
The functionality is minimal and I want to warn you not to use flash for these first versions - use SRAM to avoid getting a brick :) I warned.

yrabbit, 2023-08-28:
Bad news: TangPrimer20k support does not mean TangNano20k support, because although the series of chips (GW2A-18 and GW2AR-18C) are similar, they are a) not identical and b) engineering decisions made when designing the TangNano20k board itself will require special software perversions.
The last point so far concerns the fact that a traditional external oscillator is connected to a regular input, not a dedicated clock input, which will require some fun decisions when routing clocks.

The good news is that both TangPrimer20k and Tangnano20k are supported in the WIP version of apicula/nextpnr, which is undergoing a final haircut before being published as alpha.

To add Tang * 20k to gowin_pll, I think it will be enough to find a datasheet somewhere and write the allowable PLL frequencies into this script.

What we need is someone with the knowledge of how to program an external PLL since that is the TangNano20k's stock clock. For me, this instruction is a complete fog so far :)

https://wiki.sipeed.com/hardware/en/tang/tang-nano-20k/example/unbox.html#pll_clk

pepijndevos, 2023-08-30:
huh fun, the MS5351 PLL seems to be an I2C device, so I guess it's just configured by their BL616 fake FTDI USB serial emulation. But yea that's a whole fresh can of worms.

tmolteno, 2023-09-02:
The MS5351 is fairly complex. I have been trying to write a python PLL settings generator for this chip for a different application (a software defined radio).  I would be happy to try and help, but it's a mess (something like 188 registers need to be written via i2c to set up a new frequency) why would the external PLL need to be programmed by apycula at all?

yrabbit, 2023-09-02:
because it is soldered to normal clock wires as opposed to a simple standard external oscillator soldered to the input of a dead PLL :)
[Tang_Nano_20K_3921_Schematic.pdf](https://github.com/YosysHQ/apicula/files/12502771/Tang_Nano_20K_3921_Schematic.pdf)

tmolteno, 2023-09-03:
OK thanks for the explanation. I have some progress on automatically calculating PLL settings for the MS5351 using python [here](https://github.com/tmolteno/zl4aa/blob/main/sdr/sdr_code/src/Quadrature_Synthesis.ipynb)... There is some C code in the same project...

laurentiuduca, 2023-09-05:
they have a group on telegram https://t.me/sipeed where you can ask

tmolteno, 2023-09-06:
Thanks @yrabbit. I've done a little digging, and this what  I found...

 It appears that the BL616 listens to the external USB UART. This exposes a command-line interface on the BL616.

One of the command line tools, <code>pll_clk</code>, allows the user to set the frequency of the three MS5351 clock outputs. These 3 clock outputs  correspond to pins

- GCLK6_P
- GCLK6_N
- LPLL2_in

So I would guess that this step would be performed by the designer before starting on HDL (effectively setting the an external reference frequency, rather like choosing a crystal oscillator frequency).

Running this <code>pll_clk</code> tool could perhaps be done using some kind of script on the host that spoke to the serial port. This is not something done via JTAG, so presumably not done by the FPGA loader.  How would you envisage apicula configuring the MS5351 in a design workflow?

yrabbit, 2023-09-06:
After your explanation, no way :(
Without the ability to program it from inside the FPGA, we can only use it if it has a default generated frequency after power-up in the case of the Tangnano20k. I wonder what it is, if any? :)

tmolteno, 2023-09-06:
I don't have a TangNano20k to test this. I don't even know if it is set on startup at all, in fact I'd guess that all three MS5351 clocks are likely to be disabled by default at startup.

yrabbit, 2023-09-06:
Better and better.
Well, when the PR https://github.com/YosysHQ/nextpnr/pull/1208 is approved, life will become a little easier - we won't have to mess with the PLL to get rid of glitches, just mark the network as CLOCK :)

yrabbit, 2023-09-17:
Tangnano20k and TangaPrimer20k are supported after commit.
The issue with the “bad” connection of the clock generator in Tangnano20k has also been resolved
 https://github.com/YosysHQ/apicula/pull/199

closed 2023-09-17 by yrabbit

retpolanne, 2023-09-28:
@yrabbit that's great news and I'm so happy :) 

I have a Tangnano20k for testing, if you need volunteers.

yrabbit, 2023-09-28:
> @yrabbit that's great news and I'm so happy :)
> 
> I have a Tangnano20k for testing, if you need volunteers.

Certainly!
two options:
- simple. download an artifact with examples and try those related to tangnano20k;
- difficult. build the apicula and nextpnr master branches locally and try your design :)

https://github.com/YosysHQ/apicula/suites/16525435709/artifacts/944262835

retpolanne, 2023-10-01:
Hey @yrabbit, is GW2AR-18 supported on nextpnr? I don't see any commits related to it.

yrabbit, 2023-10-01:
R means that some memory has been added to the chip. The FPGA itself is exactly the same.

retpolanne, 2023-10-03:
Got it. I believe Tang Nano 20k itself has the R in it.
https://wiki.sipeed.com/hardware/en/tang/tang-nano-20k/nano-20k.html

yrabbit, 2023-10-03:
yeah, tangnano20k is supported by Apicula.