YosysHQ/apicula #131: Support for Gowin IDDR module

uhyonc, 2022-10-28:
I was trying to use the Open source PSRAM module (https://github.com/zf3/psram-tang-nano-9k)
However that module uses the IDDR module which doesn't seem to exist in Yosys/apicula.
Can we add support for the IDDR module? (Or maybe a pointer on how I can help implement this for apicula?)

Thanks!

yrabbit, 2022-10-28:
This primitive along with its brethren ser/deser 4, 8, 10 are planned for implementation I think not before the end of the year.
At the same time the I/O subsystem will probably be rewritten because it is currently too complex, relies heavily on manually discovered magic numbers and is poorly extensible.
I don't recommend it by any means, but maybe now the best option would be to use a vendor IDE?

yrabbit, 2023-04-20:
IDDR

yrabbit, 2023-04-20:
IDDR and the entire IDES family is implemented