YosysHQ/apicula #136: Incorrect chipdb version ?

meyanik, 2022-12-08:
I'm a newbie to the project. I have a Tang-nano1k board and trying to flash a basic led=button project. But when I am using makefile which i wrote using example makefile i got this error:
" ERROR: Incorrect chipdb version 1 is used. Version 2 is required "
I think device name should be different. I am using GW1NZ-LV1QN48C6/I5
my makefile line:
" $(NEXTPNR) --json $< --write $@ --family GW1NZ-1 --cst tangnano1k.cst --device GW1NZ-LV1QN48C6/I5 "
any ideas how can i fix it ?

pepijndevos, 2022-12-08:
Update apycula to 0.5.1

closed 2022-12-08 by pepijndevos

meyanik, 2022-12-08:
After installing Apycula-0.5.1 over pip, i get the same error unfortunately :(

yrabbit, 2022-12-08:
I don't know much about installations, but let's try to work it out:
- This error is printed by nextpnr;
- The bases, which are not external files, are embedded in nextpnr during the compilation process;

You probably have a creepy version of nextpnr which has the master branch code, but the bases were used from an older version of apicula when building it.

If you compiled nextpnr yourself, the sequence may help:
install the latest apicula -> compile nextpnr -> compile example

in theory:)

mmicko, 2022-12-09:
@meyanik other option is to use prebuilt oss-cad-suite from  https://github.com/YosysHQ/oss-cad-suite-build/releases/latest

meyanik, 2022-12-10:
Thank you all, after removing and re-installing apycula with pip and pre-compiling nextpnr  bba.cmake files and installing nextpnr, i got it worked.