YosysHQ/apicula #143: Stopped triggering on `posedge clk/52` in the 2022-12-16 release

HCanber, 2022-12-19:
I'm totally new to this so forgive me if I don't use the correct terminology (and feel free to rephrase the title to make it clearer :)

I have this:

```verilog
module counter
(
    input clk,
    output [5:0] led
);

localparam WAIT_TIME = 13500000;
reg [5:0] ledCounter = 4;
reg [23:0] clockCounter = 0;

always @(posedge clk) begin
    clockCounter <= clockCounter + 1;
    if (clockCounter == WAIT_TIME) begin
        clockCounter <= 0;
        ledCounter <= ledCounter + 1;
    end
end

assign led = ~ledCounter;
endmodule
```

and this `.cst` file:
```
IO_LOC  "clk" 52;
IO_PORT "clk" IO_TYPE=LVCMOS33 PULL_MODE=UP;

IO_LOC  "led[0]" 10;
IO_PORT "led[0]" DRIVE=8 IO_TYPE=LVCMOS18;

IO_LOC  "led[1]" 11;
IO_PORT "led[1]" DRIVE=8 IO_TYPE=LVCMOS18;

IO_LOC  "led[2]" 13;
IO_PORT "led[2]" DRIVE=8 IO_TYPE=LVCMOS18;

IO_LOC  "led[3]" 14;
IO_PORT "led[3]" DRIVE=8 IO_TYPE=LVCMOS18;

IO_LOC  "led[4]" 15;
IO_PORT "led[4]" DRIVE=8 IO_TYPE=LVCMOS18;

IO_LOC  "led[5]" 16;
IO_PORT "led[5]" DRIVE=8 IO_TYPE=LVCMOS18;
```

Building and loading this on a Tang Nano 9k (`make load`) worked fine using the [2022-12-15 release](https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2022-12-15). The leds are changing as `ledCounter` is increased.

However, if I build and load this using the [2022-12-16 release](https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2022-12-16) it loads correctly, but the leds are _not_ changing. One led is turned on (as `ledCounter` is initiated with 4). 
It's like the whole `always posedge clk` block isn't there.

Comparing the build files i noticed the generated `_pnr.json` files differs between versions, and of course the bitstream differs.

I'm running the darwin-arm64 versions.

Here is the source code and generated files for each release version
[counter_2022-12-15.zip](https://github.com/YosysHQ/apicula/files/10262412/counter_2022-12-15.zip)
[counter_2022-12-16.zip](https://github.com/YosysHQ/apicula/files/10262416/counter_2022-12-16.zip)

Let me know if you need any more info.

pepijndevos, 2022-12-19:
Thanks for the report. Based on your code I expect it is a known issue that can be worked around by passing `-nodffe` to yosys.

lushaylabs, 2022-12-19:
I tried it with `-nodffe` as-well as `-noalu` it doesn't solve this one, looks like a new issue. 

I simplified the example to a single led and a smaller counter and it still reproduces the issue:
```verilog
module top
(
    input clk,
    output reg led = 1
);

reg [1:0] counter = 0;

always @(posedge clk) begin
    counter <= counter + 1;
    if (counter == 2'b11) begin
        led <= ~led;
        counter <= 0;
    end
end

endmodule
```

I unpacked the generated fs file for this shortened example, one was generated with OSS-Cad-Suite 2022-12-19 (Apicula 0.6) and one with OSS-Cad-Suite 2022-12-15 (Apicula 0.5.2). I am not that familiar with the internals so I am not sure what I am looking for but maybe the images / dot files can help you understand why this might be happening.

Dot file for version 2022-12-15
[image]

Dot file for version 2022-12-19
[image]

I unpacked both fs files with the same older version of yosys and apicula for consistency, let me know if that makes a difference I can generate some new ones.

[dotfiles.zip](https://github.com/YosysHQ/apicula/files/10263414/dotfiles.zip)

yrabbit, 2022-12-20:
if it's not too much trouble, can you send the .fs and -pnr.json of the little example that doesn't work? 
And to rule out randomness, generate the .fs and -pnr.json with the same --seed value, please. Like --seed=10

yrabbit, 2022-12-20:
With the patch, both your big counter and the small counter work.
When generating the base, non-existent clock wires were created, which nextpnr used, but the resulting image of course could not work.
[image]

https://user-images.githubusercontent.com/6075465/208582374-1ec095ae-65ee-4e0f-834a-4cbd80ac70fe.mp4

closed 2022-12-20 by pepijndevos