YosysHQ/apicula #155: Cannot programm TangNano wrong id-code

KozikR, 2023-01-21:
I set 
DEVICE to GW1N-LV1QN48C6/I5
BOARD to tangnano
and create pack.fs from blink example.

When I use GOWIN programmer version 1.9.8.10 got error:
[image]

trabucayre, 2023-01-21:
There is two tangNano 1k: one based on an **GW1N-1** (the original) and one (the new revision) with a **GW1NZ-1**
Could you check idcode from your board ?:
- **GW1NZ-1**: `0x0900281B`
- **GW1N-1**: `0x0100681B`

KozikR, 2023-01-21:
ID code is

Info:	Cable found:  USB Debugger A/0/10545/null (USB location:10545)
Info:	Cost 0.1 second(s)
:	 
Info:	Target Cable: USB Debugger A/0/10545/null@2.5MHz
Info:	Operation "Read Device Codes" is starting on device-1...
Info:	ID code is: **0x0900281B**
Info:	User code is: 0x0000496F
Info:	Status code is: 0x0001F020
Info:	Cost 0.54 second(s)

So looks like GW1NZ-1 but on chip package there is GW1N-LV1QN48C6/I5:

[image]

I change DEVICE to GW1NZ-LV1QN48C6/I5 end rerun:
```
nextpnr-gowin --json blinky.json --write pnrblinky.json --device $DEVICE --cst $BOARD.cst
gowin_pack -d $DEVICE -o pack1.fs pnrblinky.json
```
But got simmilar issue:
[image]

trabucayre, 2023-01-21:
My fault: I've inverted idcode... It's a GW1N-1. Sorry.
Could you share your .fs (the one targeted for GW1N-LV1QN48C6/I5) ?
Thanks

KozikR, 2023-01-21:
[pack.zip](https://github.com/YosysHQ/apicula/files/10472914/pack.zip)

trabucayre, 2023-01-21:
the idcode contained into pack.fs is okay. I have tested your file with my tangnano: it works (but haven't used proprietary tool).
Seems the issue isn't apicula related but due to the programmer...

yrabbit, 2023-04-20:
At the moment both tangnano (GW1N-1) and tangnano1k (GW1NZ-1) flash without problems.

closed 2023-04-20 by yrabbit

marsohod4you, 2023-09-07:
I met similar issue that Gowin native programmer refuses to use bitstream file after gowin_pack.
It seems that native gowin programmer needs header like that:

//Copyright (C)2014-2022 Gowin Semiconductor Corporation.
//All rights reserved.
//File Title: Bitstream file
//GOWIN Version: V1.9.8.09
//Part Number: GW1NR-UV9QN88PC6/I5
//Device-package: GW1NR-9-QFN88P
//BackgroundProgramming: OFF
//CheckSum: 0xCB1A
//UserCode: 0x0000CB1A
//SPIAddr: 0x00000000
//LoadingRate: 2.500MHz
//CRCCheck: OFF
//Compress: OFF
//Encryption: OFF
//SecurityBit: ON
//SecureMode: OFF
//JTAGAsRegularIO: OFF
//Created Time: Thu Sep 07 17:44:22 2023

I added this header at the begin of gowin_pack generated bitstream and then could use Gowin native 
programmer.

pepijndevos, 2023-09-08:
I am not opposed to generating a header in apicula, but couldn't get their programmer to work at all.

Would you be interested to make a pr for this?
I expect not all headers are required, so it'd take a bit of experimentation which parts are required and how to fill them in from apicula