YosysHQ/apicula #157: gowin_pack fails on nano4k on hdmi code with strange IOB15A and IOB15B error

martell, 2023-01-29:
I've been trying to build my_hdmi_device by splinedrive with apicula and have gotten pretty far along with some tweaks.
gowin_pack with the current apicula/chipdb fails with these 2 key errors on the nano4k

```
gowin_pack -d GW1NS-4 -o chip_balls.fs chip_balls_pnr.json
Traceback (most recent call last):
  File "/Users/martell/dev/oss-cad-suite/lib/python3.8/site-packages/Apycula-0.6.2-py3.8.egg/apycula/gowin_pack.py", line 435, in place
  File "/Users/martell/dev/oss-cad-suite/lib/python3.8/site-packages/Apycula-0.6.2-py3.8.egg/apycula/chipdb.py", line 941, in loc2bank
KeyError: 'IOB15B'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/martell/dev/oss-cad-suite/libexec/gowin_pack", line 33, in <module>
    sys.exit(load_entry_point('Apycula==0.6.2', 'console_scripts', 'gowin_pack')())
  File "/Users/martell/dev/oss-cad-suite/lib/python3.8/site-packages/Apycula-0.6.2-py3.8.egg/apycula/gowin_pack.py", line 703, in main
  File "/Users/martell/dev/oss-cad-suite/lib/python3.8/site-packages/Apycula-0.6.2-py3.8.egg/apycula/gowin_pack.py", line 439, in place
Exception: IOB15A is not allowed for a given package
make: *** [chip_balls.fs] Error 1
```

Steps to reproduce

```
git clone https://github.com/martell/my_hdmi_device
cd my_hdmi_device
git checkout yosys-nano4k
make -f Makefile.nano4k
```

@yrabbit @pepijndevos This seems very close since merging #154

yrabbit, 2023-01-30:
At the end of the output Yosys shows that you did not use the TLVDS_OBUF primitive, without using it nextpnr considers the wire pair separate and may assign a random pin to the negative wire, which may not be allowed in this particular case.
This, of course, should somehow be detected earlier, but it won't solve your problem.
[image]

This is how the Yosys output ends up in the tlvds-tangnano4k example. 
[image]

martell, 2023-01-30:
I am definitely out of my depth here but but my understanding was that you could use ODDR and leave Q1 floating instead of using TLVDS for Double data rate instead of Single data rate.
See here https://github.com/splinedrive/my_hdmi_device/blob/master/chip_balls.v#L413
When I comment out the ODDR lines and uncomment the TLVDS usage it builds like you said.
Does this means the vendor compiler supports this while yosys does not ?

Since yesterday I changed this line to reduce the LUT usage
https://github.com/splinedrive/my_hdmi_device/commit/a8c325bc882e388de3ad519e5874029b9eb5a9f3
This leads me to believe something in yosys is causing large LUT usage (maybe yosys is not stripping unused logic? ) because the upstream project works with the gowin ide and the nano4k.

yrabbit, 2023-01-30:
Well TLVDS has nothing to do with DRR and data rate at all - it's just a way of electrical signal transmission: https://semiengineering.com/knowledge_centers/communications-io/off-chip-communications/i-o-enabling-technology/lvds-low-voltage-differential-signaling/

And when you specify two pins for the same network (IO_LOC "hdmi_p[0]" 30,29;), you let nextpnr know that you will use a pair of pins: one as +, the other as -. This is LVDS, then nextpnr expects to find the TLVDS_OBUF primitive and process it.

Yes it is bad and we should swear at such a thing and we will do it in future.

closed 2023-01-30 by martell