YosysHQ/apicula #158: gowin_pack fails on nano9k with strange LVCMOS18 error

martell, 2023-01-30:
I've been trying to build my_hdmi_device by splinedrive with apicula and have gotten pretty far along with some tweaks.
gowin_pack with the current apicula fails with this strange error on the nano9k

```
gowin_pack -d GW1N-9C -o chip_balls.fs chip_balls_pnr.json
Traceback (most recent call last):
  File "/Users/martell/dev/oss-cad-suite/libexec/gowin_pack", line 33, in <module>
    sys.exit(load_entry_point('Apycula==0.6.2', 'console_scripts', 'gowin_pack')())
  File "/Users/martell/dev/oss-cad-suite/lib/python3.8/site-packages/Apycula-0.6.2-py3.8.egg/apycula/gowin_pack.py", line 703, in main
  File "/Users/martell/dev/oss-cad-suite/lib/python3.8/site-packages/Apycula-0.6.2-py3.8.egg/apycula/gowin_pack.py", line 452, in place
Exception: Different I/O modes for the same bank 1 were specified: LVCMOS18 and LVCMOS33
make: *** [chip_balls.fs] Error 1
```

Steps to reproduce

```
git clone https://github.com/martell/my_hdmi_device
cd my_hdmi_device
git checkout yosys-nano9k
make -f Makefile.nano9k
```

@yrabbit @pepijndevos This seems very close since merging #156

yrabbit, 2023-01-30:
Let's see if it doesn't get resolved after we beat 

https://github.com/YosysHQ/apicula/issues/157

martell, 2023-01-30:
Continuing from there trying to use LVDS here this is where we get stuck

I thought the nano9k had emulated lvds instead of true from reading docs and this reddit thread proved useful.
https://www.reddit.com/r/GowinFPGA/comments/wu96b5/difference_between_gowin_true_and_emulated_lvds/

Using ELVDS_OBUF I get the following error - I presume apicula just doesn't have support for it yet ?
```
7.3.21. Executing AST frontend in derive mode using pre-parsed AST for module `\ball'.
Generating RTLIL representation for module `$paramod$df58ad4b67c0055eee254f8d5891cca82e58fb35\ball'.
ERROR: Module `\ELVDS_OBUF' referenced in module `\chip_balls' in cell `\clk_tlvds_obuf' is not part of the design.
```

Using TLVDS gowin pack errors out with this
```
Exception: IOT42A. TLVDS_OBUF is not allowed for a given io standard LVCMOS33
```

I presume some logic has to be added to apicula for ELVDS_OBUF which should work with LVCMOS33 ?
I'd love to look at implementing this if you can point me in the right direction (assuming it is not too complicated)

yrabbit, 2023-01-30:
Oh shit, I totally forgot about this feature of the Tangnano9k board - there are TLVDS pins, but they are not outputted to the HDMI connector. Instead, the LVDS emulation is put there.
And we don't support this primitive yet if I'm not mistaken :(

yrabbit, 2023-01-30:
I don't think changing the standard will help: TLVDS primitives are hardwired to their respective pins and cannot be placed on the HDMI pins on that particular board.

yrabbit, 2023-01-30:
I don't know your qualifications and can't judge whether it's difficult or not:
- To add the recognition of the ELVDS primitive proper to yosys;
- add the creation of such Bels in apycula bases;
- teach nextpnr how to properly place and route such things;
- add to apycula bases what bits should be set for given ELVDS parameters.

To me, this is a pain in the ass.
But I will do it after PLL anyway.

yrabbit, 2023-01-30:
The vendor tools on your 4k project also show good use of LUT and DFF :)
Here we are clearly losing out on optimization.
[image]

yrabbit, 2023-05-29:
ELVDS family primitives added by https://github.com/YosysHQ/apicula/pull/177

closed 2023-05-29 by yrabbit