YosysHQ/apicula #167: Hardware DSP support for GW1N-9C?

kehribar, 2023-02-16:
Hi,

It seems like apicula currently _doesn't_ support the hardware DSP capabilities [UG287E.pdf] [1] of Gowin FPGA chips.

I'm not currently interested in automatic inference of DSP modules based on higher level Verilog code but I'm looking for a way to manually instantiate MULT18X18 blocks etc.

Am I missing anything?

Thanks a lot for your efforts.

[1]: http://cdn.gowinsemi.com.cn/UG287E.pdf

[image]

pepijndevos, 2023-02-17:
It is indeed not supported but on the roadmap. If you'd like to help out, come hang out in matrix/irc :)

yrabbit, 2024-04-08:
You can try the latest release of apicula.

kehribar, 2024-04-16:
Thanks for the update! I'll close this issue for now.

closed 2024-04-16 by kehribar