YosysHQ/apicula #172: bitstream has 0x1100581B hardware requires 0x0100481b

lukaspfitscher, 2023-04-07:
yosys -D LEDS_NR=6 -p "read_verilog blinky.v; synth_gowin -json blinky.json"
nextpnr-gowin --json blinky.json \
              --write pnrblinky.json \
              --device GW1NR-LV9QN88PC6/I5 \
              --cst tangnano9k.cst
gowin_pack -d GW1NR-LV9QN88PC6/I5 -o pack.fs pnrblinky.json
openFPGALoader -b tangnano9k pack.fs

ERROR:
openFPGALoader -b tangnano9k pack.fs
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Parse file Parse pack.fs: 
Done
DONE
Error: Failed to claim FPGA device: mismatch between target's idcode and bitstream idcode
	bitstream has 0x1100581B hardware requires 0x0100481b

Is it a bug or did i something wrong? i use OSS CAD Suite

yrabbit, 2023-04-07:
GW1NR-9 and GW1NR-9C are different chips, but they have the same model number. 
So that's the trouble.
You make an image for 9 and try to flash it to 9C.
As a quick fix, you can try it:
`gowin_pack -d GW1N-9C .......`

lukaspfitscher, 2023-04-07:
Thanks it works!

Should we add this info somewhere, from a beginner like me, it's hard to find these things out

Here is the updated command:
yosys -D LEDS_NR=6 -p "read_verilog blinky.v; synth_gowin -json blinky.json"
nextpnr-gowin --json blinky.json \
--write pnrblinky.json \
--device GW1NR-LV9QN88PC6/I5 --family GW1N-9C \
--cst tangnano9k.cst
gowin_pack -d GW1N-9C -o pack.fs pnrblinky.json
openFPGALoader -b tangnano9k pack.fs

closed 2023-04-07 by lukaspfitscher