YosysHQ/apicula #185: why you guys so powerful

quantrpeter, 2023-07-09:
hi, gowin fpga seems not open source, why you guys so powerful to generate the bitstream file? How can you do that? thanks

yrabbit, 2023-07-09:
It's simple: you take an empty file
```
module top();
endmodule

```
generate an image with the vendor IDE, this will be your template, then you add one LUT, generate the image and compare it with the template. Change the position of the LUT and see what changed and what didn't.
Take a look at the document https://github.com/YosysHQ/apicula/blob/master/clock_experiments.ipynb
or the movies: 

https://www.youtube.com/watch?v=Vt7FyOXfkZA&list=PLIYslVBAlKZad3tjr5Y4gqBV3QKQ5_tPw

pepijndevos, 2023-07-09:
Or my internship report https://github.com/pepijndevos/internshipreport/releases/download/tag-f2973b39f18d72c18a0ede2d765fb1ff8541cdeb/main.pdf