YosysHQ/apicula #187: Missing Apycula file when compiling nextpnr_gowin

philiprbrenan, 2023-07-16:
I am trying to compile nextpnr-gowin after installing apycula using this github action:

```https://github.com/philiprbrenan/nextPnr/actions/runs/5565472535/jobs/10165851576```

The compilation process terminates with the following message concerning an apycula file:

```FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.10/dist-packages/apycula/GW2A-18.pickle'```

Please tell me how to obtain this file?

yrabbit, 2023-07-16:
This file should be in the apicula 0.8.2a1 release. 
The non-standard way is to download gw2a-18-chipdb from artifacts.
https://github.com/YosysHQ/apicula/actions/runs/5311060550#artifacts

mulianov, 2023-07-17:
This leads to 
[  1%] Generating chipdb/chipdb-GW2A-18.bba
[  3%] Built target bbasm
Traceback (most recent call last):
  File "/home/user/.local/venv/bin/gowin_bba", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/user/.local/venv/lib/python3.11/site-packages/apycula/gowin_bba.py", line 258, in main
    db = pickle.load(f)
         ^^^^^^^^^^^^^^
_pickle.UnpicklingError: invalid load key, '\x1f'.

yrabbit, 2023-07-17:
apicula 0.8.2a1 release + nextpnr master.

mulianov, 2023-07-22:
@yrabbit In message above there is the result of such combination - "_pickle.UnpicklingError: invalid load key, '\x1f'."

yrabbit, 2023-07-22:
This error message indicates that there is an attempt to open the chip base zip file in the usual way as an unpackaged file.

You can see for yourself.
[image]

nextpnr uses gowin_bba from apicula installed on your computer, and if it is 0.8.2a1, it is very unlikely that it uses unpackaged database loading.

And you can find all gowin_bba.py on your computer and check that it is version 0.8.2a1 by looking at line 258.

https://github.com/YosysHQ/apicula/blob/87131a1f997eebe1d9a6b4fc2f55a211eeeb962f/apycula/gowin_bba.py#L258

closed 2023-08-30 by yrabbit