YosysHQ/apicula #188: AttributeError: module 'importlib.resources' has no attribute 'files'

marsohod4you, 2023-07-26:
I am trying to build custom project for Gowin chip. But cannot pack.
With version  oss-cad-suite-windows-x64-20230508.exe this worked correctly for my project.
But now after update with version oss-cad-suite-windows-x64-20230726.exe I get error message:
```
gowin_pack -d GW1NR-UV9QN88C6/I5 -o output\out-pack.fs output\out-pnr.json
<frozen importlib._bootstrap>:219: Warning: Numpy built with MINGW-W64 on Windows 64 bits is experimental, and only available for
testing. You are advised not to use it for production.

CRASHES ARE TO BE EXPECTED - PLEASE REPORT THEM TO NUMPY DEVELOPERS
Traceback (most recent call last):
  File "d:\Gowin\oss-cad-suite\bin\gowin_pack-script.py", line 33, in <module>
    sys.exit(load_entry_point('Apycula==0.8.2a2.dev1+g8e31050', 'console_scripts', 'gowin_pack')())
  File "D:\Gowin\oss-cad-suite\lib\python3.8\site-packages\Apycula-0.8.2a2.dev1+g8e31050-py3.8.egg\apycula\gowin_pack.py", line 984, in main
AttributeError: module 'importlib.resources' has no attribute 'files'
```
What can be issue with new oss-cad-suite?

yrabbit, 2023-07-26:
It's all about the Python version. The current state of affairs in this area can be obtained from the persons involved in the conversation, I think:)

https://github.com/YosysHQ/apicula/pull/183#issuecomment-1639125762

pepijndevos, 2023-07-26:
Hm no news from our new friend right? I'll look at it next week when I'm back home.

marsohod4you, 2023-07-27:
> It's all about the Python version. The current state of affairs in this area can be obtained from the persons involved in the conversation, I think:)

I was thinking oss-cad-suite uses internal python distro. Is it correct? So my additional pythons installed in system cannot be trouble.
At least with previous oss-cad-suite I had near installation python 3.9 and all worked.
Now I have uninstalled all python versions I had on my Windows PC and I see that still yosys and nextpnr-gowin work properly only gowin-pack fails.

Should I try to install in my Windows OS additional Python 3.8? I think not..

sergeyu, 2023-08-02:
Having the same issue on Mac arm64 (M2 Max).

pepijndevos, 2023-08-03:
fixed by #189

closed 2023-08-03 by pepijndevos