YosysHQ/apicula #195: GW1NZ-1 rPLL (nano 1k) does not synthesize working pll.

dirkenstein, 2023-08-24:
Have tried several PLL configs which all produce locking/working PLLs in gowin tools but exactly the same verilog results in a PLL that doesn’t lock and has no output.  All on Tang nano 1k, input is 27MHz external clock, output varies between 33M and 100M.

yrabbit, 2023-08-24:
The Z-1 is a fun chip, it seems to have an interesting power-saving system that is not found anywhere else that turns off the power to the PLL by default.
This thing I think is fixed in the wip version of apicula, but not in the current one.

this file is apycula/gowin_pack.py, if you are feeling adventurous you can add these lines. Well, or write to me - I'll do it and make a gowin_pack for you.

```
   dualmode_pins(db, tilemap, args)
+    # XXX Z-1 some kind of power saving for pll, disable
+    if device in {'GW1NZ-1'}:
+        tile = tilemap[(db.rows - 1, db.cols - 1)]
+        for row, col in {(23, 63)}:
+            tile[row][col] = 0

    res = chipdb.fuse_bitmap(db, tilemap)
    header_footer(db, res, args.compress)
    if pil_available and args.png:
```

dirkenstein, 2023-08-25:
Yup, that fixes it. Do you want the needed updates to gowin_pll for GN1NZ-1 use as well?

yrabbit, 2023-08-25:
It would be priceless!

dirkenstein, 2023-08-25:
This should do it:

`--- gowin_pll-old.py	2023-08-19 01:15:27.000000000 +0100
+++ gowin_pll.py	2023-08-19 23:13:36.000000000 +0100
@@ -64,6 +64,16 @@
             "clkout_min": 2.5,
             "clkout_max": 360,
         },
+        "GW1NZ-1 C6/I5": {
+            "comment": "tested on TangNano1K Board",
+            "pll_name": "rPLL",
+            "pfd_min": 3,
+            "pfd_max": 400,
+            "vco_min": 400,
+            "vco_max": 800,
+            "clkout_min": 3.125,
+            "clkout_max": 400,
+        },
         "GW1NR-2 C7/I6": {
             "comment": "Untested",
             "pll_name": "PLLVR",
`

yrabbit, 2023-08-25:
Can you do PR? Then your authorship will be inscribed in a normal way in history

dirkenstein, 2023-08-25:
Will do, can’t do it before Monday

yrabbit, 2023-08-30:
Should be fixed in https://github.com/YosysHQ/apicula/pull/184

closed 2023-08-30 by yrabbit