YosysHQ/apicula #200: Question - how to include last commits of apicula in nextpnr

laurentiuduca, 2023-09-18:
Hi,

  I have seen that now apicula with last commits support gw2a-18 of tang nano 20k.
I did not find a how to include the version of apicula with these commits in nextpnr compilation. So, please give us instructions of how to do this.

Thank you

pepijndevos, 2023-09-18:
The most up to date on how to build apicula and nextpnr is always https://github.com/YosysHQ/apicula/blob/master/.github/workflows/chipdb.yml

But yeah we should update the readme once all the changes have stabilized a bit.

yrabbit, 2023-09-18:
The brave may look at the line in the CI file, but this is at their own peril and risk and no consultation until the official instructions in the README ;)

https://github.com/YosysHQ/apicula/blob/cff0cd80654652b63ff2f48daed3b4956e6da12b/.github/workflows/chipdb.yml#L305

laurentiuduca, 2023-09-19:
nextpnr$ cmake . -DBUILD_PYTHON=OFF -DARCH="gowin;himbaechel" -DHIMBAECHEL_GOWIN_DEVICES="GW1N-1;GW1NZ-1;GW1N-4;GW1N-9;GW1N-9C;GW1NS-4;GW2A-18;GW2A-18C" -DPython3_EXECUTABLE=${{ steps.pysetup.outputs.python-path }} 
bash: -DPython3_EXECUTABLE=${{ steps.pysetup.outputs.python-path }}: bad substitution

yrabbit, 2023-09-19:
you don't have to copy this letter by letter - this is just an example of how to compile nextpnr

pepijndevos, 2023-09-20:
To be more specific, remove the whole ` -DPython3_EXECUTABLE=${{ steps.pysetup.outputs.python-path }}` since it is just to tell CI where to find Python. Just make sure you use the same python binary for Apicula and nextpnr and you should be good.

yrabbit, 2024-01-27:
Lost relevance with the release of nextpnr with himbaechel-gowin

closed 2024-01-27 by yrabbit