YosysHQ/apicula #203: FPGA initialises but doesn't run. Same code works on the vendor tools.

hugoferreira, 2023-10-11:
Now, this is a pretty sanity check that is driving me insane (pun intended).

Tang Nano 4k. Extremely simple code:

```
module led_water (
    input clk,
    input rstn,
    output reg led_o
);

reg [23:0] counter;

always @(posedge clk or negedge rstn) begin
    if (!rstn)
        counter <= 24'd0;
    else if (counter < 24'd675_0000)
        counter <= counter + 1;
    else
        counter <= 24'd0;
end

always @(posedge clk or negedge rstn) begin
    if (!rstn)
        led_o <= 1'b1;
    else if (counter == 24'd675_0000)
        led_o <= ~led_o;
end

endmodule
```

The result of this through Yosys + Nextpnr + OpenFPGA (board "GW1NSR-LV4CQN48PC7/I6" matching the chip markings) results in a non-blinking LED. The same code using GoWin IDE (running Wine), blinks the led as expected.

Using OpenFPGA with the bitstream generated by the vendor tools _also_ works. I haven't yet tried synthesising using the vendor tools and then routing using apicula (is this possible?) But either I am messing something _extremely_ fundamental, or?...

yrabbit, 2023-10-11:
You can also try:
0) use an external LED (via a resistor, of course)
1) use himbaechel architecture for gowin: you will have to install master apicula and master nextpnr

https://github.com/YosysHQ/apicula/assets/6075465/acee16b1-cc68-46da-90a7-d473333a7f7a

marsohod4you, 2023-12-15:
I met similar issue as @hugoferreira 
Can You please clarify what You mean with "use himbaechel architecture for gowin: you will have to install master apicula and master nextpnr"?
I have just downloaded recent nightly build of oss-cad-suite from https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2023-12-14 to my Ubuntu 20.04
I have cloned apicula and trying to build with oss-cad-suite simple blinky.v for tangnano9k board
I follow instruction https://github.com/YosysHQ/apicula/blob/master/readme.md

Resulting fs file I upload to board by using native Gowin Programmer.

Project starts on tangnano9k, but not all LEDs blink from 6 LEDs only 3 LEDs seem blnking properly as binary counter, others are dark.

I use yosys, and nextpnr-gowin and gowin-pack from oss-cad-suite.
Should I try to build them all myself from master branch? Why should I do that if oss-cad-suite is built every day with new tagged release?

yrabbit, 2023-12-15:
"Why should I do that if oss-cad-suite is built every day with new tagged release?"
There is a page that, among other things, shows which version of nextpnr there are no problems with. When you talk about tagged release, you mean nextpnr version 0.6 from May 16th, right? This version does not know anything about himbaechel-apicula and it will be difficult to compile examples for this architecture there. But master has no problems, that's what I meant.
https://github.com/YosysHQ/apicula/actions/runs/7075900449#artifacts

[image]
In addition, there is an archive of already compiled examples on this page, there are ready-made images with different things. Try it from the himbaechel catalog - I wonder if all the LEDs on Tangnank9k will blink.

yrabbit, 2023-12-15:
Since I finally broke my copy of Tangnano9k and am waiting for a replacement at the end of December, I can’t fix anything now.

dciangot, 2024-01-17:
FYI I'm experiencing the same kind of issue as @hugoferreira initially reported, but with tangPrimer20k board. Has anyone tried with success the blinky.v example on primer20k ? In case, I would be glad to try replicating the very same setup. Atm I compiling both nextpnr and apicula from main branches. (also with 0.10 apicula, same issue)

yrabbit, 2024-01-17:
examples/himbaechel/blinky.v included with apicula works.
configuration apicula (master) nextpnr(master)
If you need to try exactly the one in the first post, then I’ll compile it tomorrow and take a look.

https://github.com/YosysHQ/apicula/assets/6075465/261465ec-c1f2-4cf1-a979-3be0e7bc557c

yrabbit, 2024-01-17:
I tried the option from the very first message. Again master apicula and master nextpnr. I made a repository with this example https://github.com/yrabbit/led-water
Is this how this led-water is supposed to work?
Primer 20k:

https://github.com/YosysHQ/apicula/assets/6075465/aee38bf7-f8e4-4cd2-a7a5-a7d6108e77bd

Tangnano9k:

https://github.com/YosysHQ/apicula/assets/6075465/7adca1df-d2db-4d79-ac4f-14bcf3448024

Tangnano4k (although I am desperately against using LED on the board, an external one is better):

https://github.com/YosysHQ/apicula/assets/6075465/2043a897-46a3-40f4-af9a-9a8e0c0faa07

hugoferreira, 2024-03-30:
@yrabbit is absolutely right. Once I switched from `nextpnr-gowin` to `nextpnr-himbaechel`, which can be installed by [following the instructions here](https://github.com/YosysHQ/nextpnr?tab=readme-ov-file#nextpnr-himbaechel), everything works flawlessly. I'll close this issue.

closed 2024-03-30 by hugoferreira