YosysHQ/apicula #204: mega 138k support

fduxiao, 2023-10-19:
Guys, there is a new GW5A series. Is there any plan to support it? If I want to add the support by myself, how should I get started? 

I am new to FPGA and have just known how to use your tool chains for tang nano 9k. Can you explain how you made the tool chains for it and I can then try it for mega 138k?

pepijndevos, 2023-10-19:
Of course we'd eventually like to support all Gowing FPGAs. So far the main focus has been those FPGAs for which there are widely available and affordable development boards, which tend to be in the most demand.

If you look through the repo you'll find there is my internship report on how I did the initial development, and I've also done some streams of working through issues. Another piece of useful info might be to look at previous PRs that added support for a new FPGA family.

jhol, 2023-12-20:
Cheap boards are available now: https://www.aliexpress.com/item/1005006224593018.html

pepijndevos, 2023-12-21:
Any volunteers? ;)

pepijndevos, 2024-01-01:
This will require an update of the IDE version we're using, making it more involved than just another device from an already supported series.

Luckily this is a fairly mechanical change so if anyone feels like dipping their toes in FPGA fuzzing it'd be a good place to start. Usually it involves adding a few changes to the parsers and updating the TCL scripts and the dockerfile and things like that.

Sometimes it breaks the fuzzers but we're now less dependent on them so it might be a good opportunity to get rid of "legacy" nextpnr-gowin stuff.

fduxiao, 2024-10-07:
I tried reverse engineering and got stuck for a while. I guess I'd better to know the structure by fuzzer before reverse engineering it.

I understand you figure out the bitstream format by making a template from an empty module, comparing it with other generated bitstream. But, is there any concrete step to do that? I mean what final result should I give and how is it generated?

My progress is that I figured out most cli/gui tasks are made in `libGWTE.so`. The lib provides a `TE::DesignManager` to manage design files and options. Then, it passes those to `TE::FlowManager` with data stored in a `TE::_flow_manager_private`, which holds those `syn/pnr` as task processes. The one for `pnr` is called `TE::PnrProcess`, which runs `pnr`. I noticed that it then treats `GW5A-138` separately with `libPnRProxy.so` while using `libGaoProxy.so` for other devices. Those two shared objects then call `libpnr.so`.

The good news is that it only relies on standard c/c++ library. The bad news is that tons of functions have to be understood.

You can use the netlist file (*.vg) generated by yosys, so we only have to figure out how `libpnr.so` works. My plan is to first figure out the structure of design files and options they parsed to it and then the pnr process.

Seyviour, 2024-10-07:
Hi, there's some work ongoing to get support for the GW5A devices. It's not completely up to date right now, but you can have a look at this draft [pull request](https://github.com/YosysHQ/apicula/pull/273) by @AwooOOoo.

yrabbit, 2025-06-19:
I wonder what's inside these consoles: linux with emulator or an exact “hardware” copy in fpga?

ohault, 2025-11-29:
Hi @fduxiao, can you please rename the title of this issue to something similar to https://github.com/YosysHQ/apicula/issues/419 and some references to some existing boards and chip in the Arora V series ?

yrabbit, 2025-11-30:
> The one I purchased is the Tang Mega 138K. (Labeled as NEO on board itself) I have the B version which is a few months older and does not have SSRAM. The newer C version just released recently does include SSRAM.

That's great! At least it will be possible to test the developments on version B :)

I have the C version and I'm messing around with it.

yrabbit, 2025-11-30:
Github is not the best place for long conversations. Do you have access to 

https://matrix.to/#/#apicula:matrix.org
or is it blocked by a firewall?
At least there I can say, to everyone's amusement, that I don't know what an SVF file is :)
As for being stuck, I'm not publishing the nextpnr part yet because there are too many peripherals on the docking board and the IO may be configured incorrectly, which could lead to overheating and board burnout. But I can upload the nextpnr version for you (not right now, though) if you feel lucky and are willing to risk the board :)