YosysHQ/apicula #208: Tang nano 20k support

jeremyherbert, 2023-11-15:
Hi, 

Is anyone working on support for the GW2AR-LV18QN88C8/I7 which is used on the [Tang Nano 20k](https://wiki.sipeed.com/hardware/en/tang/tang-nano-20k/nano-20k.html) ?

I suspect that these two issues may be related:

https://github.com/YosysHQ/apicula/issues/206 (NR is the same as N, but with SDRAM)
https://github.com/YosysHQ/apicula/pull/181 (Same part family, no SDRAM and different package)

Just checking here before I start trying to have a look at it myself.

yrabbit, 2023-11-15:
Tangnano20k is supported up to the point where it runs this RISC thing called ATTOSOC, look at the Makefile for examples:
https://github.com/YosysHQ/apicula/blob/a4f04e0c481d0ab5feb828a311cca1a76c7b18b6/examples/himbaechel/Makefile.himbaechel#L9-L13

If you want to put things in order with R and NR, then PR is always welcome.

jeremyherbert, 2023-11-15:
Thanks for the tip, I was looking in the examples folder only, not the himbaechel one. 

I will have a look to see if I can work out something for the R/NR.

closed 2023-11-15 by yrabbit

pepijndevos, 2023-11-15:
We should really add it to the readme, we tell people to install from git anyway so I don't think we need to wait for a release

chili-chips-ba, 2023-12-26:
@juninho99 is stuck on this BRAM-related error. Any tips or advice on how to proceed from here?
[image]

yrabbit, 2023-12-26:
nextpnr says that there are no descriptions for DPX9 primitives in the chip database.
So I'm wondering which version of nextpnr (nextpnr-gowin or himbaechel) you are using and which version.

And I’ll say right away that legacy gowin has no memory support.

chili-chips-ba, 2023-12-26:
Thank you, that's good input, unfortunately not directly forthcoming from documentation. 
- @juninho99 is now trying it with all the latest versions, and also with himbaechel

chili-chips-ba, 2023-12-27:
BTW, the design that's giving us this trouble is a game in which CPU writes into memory, while HDMI hardware is reading from it. So, it's dual-port BRAM...
- https://github.com/chili-chips-ba/openXC7-TetriSaraj

yrabbit, 2023-12-27:
But this is exactly what follows from the documentation: Tangnano9k does not support Dual Port.

[image]

I'm a little confused by how you jump from 20k to 9k. :)

yrabbit, 2023-12-27:
> BTW, the design that's giving us this trouble is a game in which CPU writes into memory, while HDMI hardware is reading from it. So, it's dual-port BRAM...
> 
> * https://github.com/chili-chips-ba/openXC7-TetriSaraj

I must say that I am very glad that a project using BRAM has appeared - I myself am not very knowledgeable about this and it will be interesting and instructive to get your thing working. Especially if you don’t use primitives directly, but trust yosys to make the memory for you - this is generally a dark forest for me.

Unfortunately, at the moment, three days before the New Year, I can only answer the simplest questions and definitely do not have the opportunity to dive into the code. I had plans to deal with memory issues around January 7th, so don’t blame me for simple answers.

chili-chips-ba, 2023-12-27:
.... thank you, your answers are super-quick in coming, and also very informative!

As the idea for this project came from @edmundhumenberger, kudos go to him :-) 

@juninho99 is a designer based in Sarajevo, Bosnia-Herzegovina, currently deeply at sleep, and anxious to answer your question first thing tomorrow morning /* while we believe he was able to use DP RAM on TangNano9K with Gowin tools, he is definitely in the best position to provide additional detail about it */

Yes, this BRAM is indeed inferred. Here is the first case:
```
//parameter MEM_INIT_FILE = "vid_ram.mem";
(* ram_style = "block" *) reg [3:0] map_ram [0:1199];   
// memory for 40x30 position Mega Character with value between 0-15 for CharacterID
```
and, here is the second:
```
//parameter MEM_INIT_FILE = "vid_ram.mem";
(* ram_style = "block" *) reg [11:0] ram [0:12287];   
// memory for 16 characters: 4095 for 640x480; 12287 for 1280x720;
```

The PLL is obviously instantiated. Juninho99 has it all working fine with Gowin proprietary flow. He is in the process of porting this design to open-source tools. We are all very enthusiastic about the value-add you see in enabling this project for nextpnr-himbaechel, and are here to do what it takes to make it happen.

yrabbit, 2023-12-27:
Everything is possible, maybe Gowin made a mistake in the documentation: I have an edition dated May 25, 2023. If it really happened in Tangnano9k to use the DP primitive, then I will make changes in 1.5 weeks (or you yourself earlier).
I have not yet researched how yosys makes memory, so here you have the freedom to change the files https://github.com/YosysHQ/yosys/blob/master/techlibs/gowin/brams.txt and https://github.com/ YosysHQ/yosys/blob/master/techlibs/gowin/brams_map.v

[image]

chili-chips-ba, 2023-12-27:
Gowin seems to be playing some yield optimization (bug engineering) games with their low-cost GW1NR family.
https://github.com/YosysHQ/apicula/pull/210#issuecomment-1817931082

Here is what we know thus far:
1) We are with no doubt using their DP RAM on TangNano9K in **Bypass** + Read-before-Write mode
- Admittedly, the board is consistently crashing (in a way that almost looked like a short) when DPRAM is configured to **Pipeline** + Read-before-Write (@tibrahimovic has complete detail) 
2) Gowin has subsequently rechristened GW1NR-9**C** to GW1NR-9
3) ... to then claim that the DP RAM was not supported at all in the "plain-vanilla" GW1NR-9

[image]
[image]
[DS117-3.0E_GW1NR series of FPGA Products Data Sheet.pdf](https://github.com/YosysHQ/apicula/files/13775402/DS117-3.0E_GW1NR.series.of.FPGA.Products.Data.Sheet.pdf)

We are reaching out to Gowin for additional detail...

yrabbit, 2023-12-27:
You will still need to edit memory inferring - I made primitives that exist at the moment, and inferring was written in those days when primitives were called differently (roughly speaking, there was no letter B in the names).
This file can be edited without recompiling yosys, I think. correct DP -> DPB and  DPX9 ->DPX9B
For me it is located along the path /usr/local/share/yosys/gowin/brams_map.v

https://github.com/YosysHQ/yosys/blob/ea7818d31bb2533d4ecceb2ed1bcf4a22b850453/techlibs/gowin/brams_map.v#L227

https://github.com/YosysHQ/yosys/blob/ea7818d31bb2533d4ecceb2ed1bcf4a22b850453/techlibs/gowin/brams_map.v#L269

yrabbit, 2023-12-27:
These are the types of errors that I am going to look for and correct after January 7, 2024 :)

chili-chips-ba, 2023-12-27:
... even the latest proprietary tool (Gowin_**V1.9.9_x64**_win) allows instantiation of DP BRAM for TangNano9K. Granted, the _Read-before-Write_ mode is not available any more, and we are now left only with _Write-Through_ and _Normal_. For the Read side, their latest tool seems to still allow _Bypass_ and _Pipeline_.

Most importantly, the **proper part to use for TangNano9K is GW1NR-9C** -- If we build for the GW1NR-9 target, the chip on the board returns IDCODE mismatch. This is all with the latest proprietary tools from Gowin...

Juninho99, 2023-12-27:
Hi,
As @yrabbit said, I changed lines 227 and 269 in `bsram_map.v` (DP -> DPB and DPX9 ->DPX9B). Additionally, I need to `make` yosys. After that I started `synth_gowin`, but after a while I got this message "/usr/local/bin/../share/yosys/gowin/cells_sim.v:0: ERROR: Cannot find object for defparam `BLK_SEL`! " .

Is there anything else I can do with bsram_map.v or any other file?

[image]

yrabbit, 2023-12-27:
These were the general instructions on how to experiment :)
Of course, you can correct it directly in the specified source file and recompile yosys every time, but I would simply find where in the system these files are INSTALLED and the edit them there. Without compilation. This makes it much faster to try different options.
[image]

Further, inderring was written (not by me) long before Gowin renamed the primitives and therefore the DP and DPB parameters are slightly different. You will have to work to ensure that they comply with the documentation (see below about the same BLK_SEL).
[image]
This is if you have the desire to dig so deep, of course, otherwise you can always go skiing or skating or celebrate noisily and wait until I finish the festive activity and get on with it.

Juninho99, 2023-12-27:
Great. Thank you. I will definitely try something because the mountains around Sarajevo are now without snow (temperatures about 10C above zero)

yrabbit, 2023-12-28:
Well, I have a feeling that this will not be a sprint, but a marathon, because I haven’t dealt with such high-level code that is used for inferring - I’m more into primitives.
And as for NIX, it’s very unlikely, because DragonFlyBSD is not among the supported platforms, and this is my main working OS; all the others are installed in very modest virtual machines.

Juninho99, 2023-12-30:
Hello Yrabbit. I hope you enjoy your holidays. I would like to say that I tried something but without much success. After changing some things in 'bsram_map.v' and the synthesis worked, I ran into another problem, related to "cell type 'DP'". I will stop here with the file changes and await your intervention. As you said, it will be a marathon for sure.

[image]

yrabbit, 2023-12-31:
Thank you.
Look, if nextpnr is trying to find a cell of type DP, then it was told to look for it. But there are no such primitives (they existed, but have already been excluded from the Gowin documentation, etc.), there are DPB and DPX9B, look, it doesn’t work in JSON after synthesizing DP primitives and find a way to teach yosys to use the correct primitives instead.

chili-chips, 2023-12-31:
We have filed an issue with Yosys team ( @cliffordwolf ).
   https://github.com/YosysHQ/yosys/issues/4098