YosysHQ/apicula #211: CLKDIV support

martell, 2023-12-16:
Hey all,

I've spent the last few months learning a lot more about chip design and have been playing with various tangnano devices again.
Learning what PLL's actually are up to the point of looking at AXI and I spent some time playing with sky130.
I implemented a CLKDIV with a divisor of 5 in Verilog logic but it does not work with the HDMI outputs on the tangnano devices so I want to figure out support for the builtin CLKDIV. I would love to help implementing this in Apicula but still do not know where or how. It seems that @yrabbit added CLKDIV to yosys here https://github.com/YosysHQ/yosys/commit/a1dd794ff883286dcd71c73fc21a701402e1078a#diff-042d23689c0c0148c5077f589e028ef752cd058c47dd66a3defe0ea62d5aa682R1474
Side Note: This is great btw because it now has a generator instead of manually created code.
Any hints or help would be greatly appreciated for this seemingly last missing piece for HDMI support.

yrabbit, 2023-12-16:
Do you mean DVI? Because any implementation of HDMI without a license invites legal action. Almost a joke.

About the primitive: for CLKDIV it is a very heavy primitive, BSRAM is a child's walk in comparison. Because the input for it is HCLK - the same two wires that run along the sides of the chip.

But, since I don’t know your experience (what if you just need an evening to implement this thing?), the approximate plan of action is as follows:
- learn to compare images with each other in order to isolate changing places;
- find the fuses responsible for the division coefficient (it's easy);
- find MUXes that connect the input of this primitive to HCLK (hard, since these MUXes are scattered throughout the chip);
- find MUXs that connect the CLKDIV output to consumers (this can be simple if the consumer can be simple logic and may turn out to be hemorrhoids if the consumers are only IO).

And yes, it’s better to immediately forget about nextpnr-gowin: of course, you can do anything, but if something doesn’t work, then dealing with the non-himbaechel option... I’ll say for myself - there’s no point.

martell, 2023-12-16:
> Do you mean DVI? Because any implementation of HDMI without a license invites legal action. Almost a joke.

For my use-case, I am happy with DVI-D to LVDS over the pins. This is for learning purposes.
By saying HDMI I am describing the port on the tangnano device which Sipeed describe as a HDMI connector.

I'll start with the first two steps before I even get to the latter ones.
> learn to compare images with each other in order to isolate changing places

I'm not sure what you mean by this.
Does this mean the images of the fpga under a microscope of looking at the gates ?

> find the fuses responsible for the division coefficient (it's easy);

I thought one could create some simple verilog examples of CLKDIV and change values to see how it behaves.
I presume that is this part ?

> himbaechel

I see commits for this in the nextpnr repo it seems to be something about a newer api to the nextpnr framework for FPGAS with larger LUT counts. I don't know how much of this works atm.

yrabbit, 2023-12-16:
> > learn to compare images with each other in order to isolate changing places
> 
> I'm not sure what you mean by this. Does this mean the images of the fpga under a microscope of looking at the gates ?

I meant to compare the images that you flash into the FPGA. To avoid repetition:
https://github.com/YosysHQ/apicula/issues/185

> 
> > find the fuses responsible for the division coefficient (it's easy);
> 
> I thought one could create some simple verilog examples of CLKDIV and change values to see how it behaves. I presume that is this part ?

Almost. You already KNOW how it behaves, at least in theory, but the bits that make this piece of FPGA behave are unknown (and interesting).

> 
> > himbaechel
> 
> I see commits for this in the nextpnr repo it seems to be something about a newer api to the nextpnr framework for FPGAS with larger LUT counts. I don't know how much of this works atm.

Look in the examples directory. In short, everything that works in legacy gowin works, plus attosoc works on Tangnano9k, Tangnano20k and TangPrimer20k.

https://github.com/YosysHQ/apicula/tree/master/examples/himbaechel

Seyviour, 2024-02-13:
> Do you mean DVI? Because any implementation of HDMI without a license invites legal action. Almost a joke.
>
> About the primitive: for CLKDIV it is a very heavy primitive, BSRAM is a child's walk in [...]

Hello :)

I'm currently working on this.

I believe I've identified the fuses that configure the DIV_MODE, and I've confirmed that the output of CLK_DIV can be consumed by simple logic (the use case I have an interest in). All that to say that I've done the easy part, lol. Finding the muxes will be more tasking, and there's the extra complication that the output of CLK_DIV in one bank can be routed to other banks. 

I have two questions, if you'll be kind enough to answer:
- How much is currently known about HCLK itself?
- Why do you expect that finding the muxes for CLK_DIV input will be hard, given that CLKDIV seems to be in the same rows/columns that the other IO tiles are on? If we already know how to use HCLK in other IO tiles, maybe this won't be so different :crossed_fingers:

yrabbit, 2024-02-13:
Wow, that was fast! Congratulations.

In addition to what is written in the Gowin documentation, we have the locations of cells with HCLK MUXes for supported chips, as well as the ranges of cells served by each of the 4 HCLK outputs. If the signal from the CLKDIV output is supplied via HCLK, then there may not be any particular difficulties even with switching to another bank - this may have already been decided inside the MUX.

https://github.com/YosysHQ/apicula/blob/f43a0b7e423ea8c946ea18f11bd2d7529f776f4e/apycula/chipdb.py#L730

My assumptions about possible difficulties are probably related to memories of research on these same HCLKs - that the wire numbers coincide with the numbers of the watch spines, but are not connected to them, then that in 9C some HCLK wires have different numbers, but in fact one wire , then in some IO cells the CLK2 wire is actually HCLK - there was a lot of fun there.

But it’s possible that you won’t have to deal with this anymore.

yrabbit, 2024-02-29:
Great job!
Let's start from the end: minimum cover for wires is a cool thing that I dreamed of :) Considering the nature of the application, I vote for the slow option - it’s still faster than what I’m looking for with my eyes, but subconsciously I won’t trust the “approximate” ;)

Regarding the “negative” wires and their fuse, there are many mysterious places in the Gowin IDE tables and this is one of them.
I'm reading lines like "if a given source is connected to a given sink, then the following fuses must NOT be set."
If this is true, then in most cases there is no point in paying attention to the absence of these fuses - the only exception is a line for which there is no “positive pair” - this is a connection when all fuses are cleared, that is, roughly speaking, an empty chip, the default connection.

Seyviour, 2024-03-01:
Thank you!

Haha. Yes, lots of weird stuff, like the alonenode table :). About the wires, the core reason why I think it'll be a good idea to report the negative wires is so that someone like me doesn't get confused in the future. Maybe we can add an argument to the function that determines if it considers the negative wires or not? 

I agree with you that the exact solver is better in this case. It might also be a good idea for it to return the top-n solutions rather than just the best one. :thinking: 

About 'logicinfo', I'd be grateful if you could kindly clarify as myy understanding of that table might be wrong.

Given that the following entries associated with DIVMODE are in 'shortval': 
```
[15, 0, fuse1, -1, -1, -1 ... -1]
[16, 0, fuse1, -1, -1, -1 ... -1]
[17, 0, fuse1, -1, -1, -1 ... -1]
[18, 0, fuse1, -1, -1, -1 ... -1]
etc. 
```
Is my task to find a 'logicinfo' table that has entries like

```
(x, 15)
(x, 16)
(x, 17)
(x, 18) 
```

yrabbit, 2024-03-01:
Well, now you remember that there is no point in paying attention to the “negative” wires except for one single one, so maybe it will be enough to just describe it in the documentation? ;)

If you need the "alonenode" table, then you're already doing those complex things so I can retire :)

Now about logicinfo. Its structure is described in the documentation (I think), but I’ll repeat it anyway. Now it is more convenient to work not with raw tables from a dat file, but from db.logicinfo.
We look at the rows of each table as follows:

```
row_number (attribute, value)
```

that is, your numbers 15, 16, 17, 18 in the shirtval table are line_number in logicinfo:

```
...
row 15 [attr_0, val_0]
row 16 [attr_0, val_1]
row 17 [attr_2, val_4]
row 18 [attr_3, val_0]
...
```
attr is the numeric code of the primitive parameter, val is the parameter value. val can be either the direct value of the parameter (if you write inst.SOME_PARAM=4 in verilog, then val is also 4), or a code (inst.SOME_OTHER_PARAM=DISABLE, then val = some_number)

Although there are few tables (25 or so), searching there by logical parameters is a very “creative” process, but if the parameter has several values, then the situation improves significantly. You're lucky: the DIV_MODE (?) parameter has several values, which means you just need to find those tables where lines 15-18 look like:

```
15: (x, val0)
16: (x, val1)
17: (x, val2)
18: (x, val3)
```

that is, where x is the same and val is different.

pepijndevos, 2024-03-02:
@Seyviour are you on our matrix channel? We're discussing plans for the coming year, in case you would like to be more actively involved.

Seyviour, 2024-03-02:
Hello @pepijndevos. Yes, I'm on the chat. I'm following the conversation but I haven't had much to say so far.

Seyviour, 2024-03-02:
Okay, then I'll add some updating the documentation to my todo list. parse_tile reporting some negative wires and not others was confusing to me, and someone that doesn't know what's going on might have a similar experience. 

Haha. I'm leaving the "alonenode" table alone for now. Maybe that's what it means -- "Leave this table alone" :) :) 

Thanks a lot for explaining the 'logicinfo' table. 

If the row-numbers start from 0, then it seems "unknown_49" is the table:
These are lines 15 - 24. Lines 20-24 also match up with the fuses for the second HCLK on a side, so I'm pretty confident.  
```
(14, 13),
 (14, 14),
 (14, 15),
 (14, 16),
 (15, 4),
 (16, 13),
 (16, 14),
 (16, 15),
 (16, 16),
 (17, 5)
```

yrabbit, 2024-03-02:
Great!
Now that you have the logicinfo table number and attribute numbers, I would advise you to do... synthesis.

With a high probability, yosys understands the primitive itself and passes it through itself, but you will have to work with nextpnr.
It needs databases and you will generate the databases :)

You need to place Bels in the right cells. Candidates will be cells with a type that has a table of 45. There may be too many of them, then you will need to experiment and determine the fuses in which cells react to changes in the divisor - there, with a high probability, it will be possible to create Bels.

closed 2024-08-29 by yrabbit