YosysHQ/apicula #213: gowin_pack fails with "'GW1NR-UV9QN881C6/I5.pickle' must be only a file name"

tpimh, 2023-12-18:
I have used apicula with OSS CAD Suite without problems (last time in August). Now I can't get `gowin_pack` to work.

Here is an example how to reproduce the error with blinky example:

```bash
cd examples
yosys -D LEDS_NR=3 -p "read_verilog blinky.v; synth_gowin -json blinky.json"
DEVICE='GW1NS-UX2CQN48C5/I4'
BOARD='honeycomb'
nextpnr-gowin --json blinky.json --write pnrblinky.json --device $DEVICE --cst $BOARD.cst
gowin_pack -d $DEVICE -o pack.fs pnrblinky.json
```

The error message:

```
Traceback (most recent call last):
  File "/home/dima/Documents/fpga/oss-cad-suite/libexec/gowin_pack", line 33, in <module>
    sys.exit(load_entry_point('Apycula==0.10.0', 'console_scripts', 'gowin_pack')())
  File "/home/dima/Documents/fpga/oss-cad-suite/lib/python3.8/site-packages/Apycula-0.10.0-py3.8.egg/apycula/gowin_pack.py", line 1384, in main
  File "/home/dima/Documents/fpga/oss-cad-suite/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/home/dima/Documents/fpga/oss-cad-suite/lib/python3.8/importlib/resources.py", line 183, in path
    resource = _normalize_path(resource)
  File "/home/dima/Documents/fpga/oss-cad-suite/lib/python3.8/importlib/resources.py", line 61, in _normalize_path
    raise ValueError('{!r} must be only a file name'.format(path))
ValueError: 'GW1NS-UX2CQN48C5/I4.pickle' must be only a file name
```

I am using the master branch of apicula and the latest build of OSS CAD Suite (oss-cad-suite-linux-x64-20231218.tgz).

tpimh, 2023-12-18:
Thanks for the quick fix! `gowin_pack` works with this patch applied. However, now I am facing a different problem: `openFPGALoader` for some reason is not supporting my board anymore:

```
$ openFPGALoader -b $BOARD pack.fs 
empty
Error: cannot find board 'honeycomb'
```

When I run `openFPGALoader --list-boards`, honeycomb is not in the list.

tpimh, 2023-12-18:
This is most likely my own mistake: previously I have flashed this board with `openFPGALoader -b littleBee -c ft2232 pack.fs`. Also worked now 😃

closed 2023-12-18 by pepijndevos

yrabbit, 2023-12-18:
I must note that you were lucky that the problem was global; if it concerned any features of the NS-2 specifically, it would have been more difficult - I don’t have this board and Gowin methodically erases all mentions of this unfortunate family from its documentation.
For example, everything is already bad with the clock:
[image]