YosysHQ/apicula #215: Simple blinky does not work on tangnano9k

marsohod4you, 2023-12-19:
Trying to build blinky example for Tangnano9k. But it does not work properly.
Board has 6 LEDs, they blink but not all: LED[1] and LED[2], LED[4] are always dark. Other LEDs blink probably properly.
When building on Gowin IDE all LEDs run properly.

What I do step by step :
0) cd apicula/examples & mkdir out
1) yosys -D LEDS_NR=6 -p "read_verilog blinky.v; synth_gowin -json out/blinky.json"
2) nextpnr-gowin --json out/blinky.json --write out/pnrblinky.json --device GW1NR-LV9QN88PC6/I5 --family GW1N-9C  --cst tangnano9k.cst
3) gowin_pack -d GW1N-9C -o out/pack.fs out/pnrblinky.json

Then I use Gowin native programmer to write "SRAM Program".
As I wrote, not all leds blink.
All above I was trying with oss-cad-suite from https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2023-12-19 under Ubuntu (most recent release)

Next my experiment was rather long.
I had built all Apicula project myself and then built Nextpnr project myself on my Ubuntu.
All from master branches 19 december 2023. 
Now I have also nextpnr-himbaechel which is missed in recent oss-cad-suite.

I was trying to build blinky.v for Tangnano9k using himbaechel (1st and 3rd steps are same as before but 2nd is different):
1) yosys -D LEDS_NR=6 -p "read_verilog blinky.v; synth_gowin -json out/blinky.json"
2) ../../nextpnr/build/nextpnr-himbaechel --json out/blinky.json --write out/pnrblinky.json --device GW1NR-LV9QN88C6/I5 --vopt family=GW1N-9C --vopt cst=tangnano9k.cst
3) gowin_pack -d GW1N-9C -o out/pack.fs out/pnrblinky.json

Write FS to board - result is exactly same - not all LEDs are blinking.
What I am doing wrong?

yrabbit, 2023-12-19:
What is the result of running these images for you?
And please send out/blinky.json, out/pnrblinky.json, out/shift.json and out/pnrshift.json (Himbaechel ones)

[images.zip](https://github.com/YosysHQ/apicula/files/13721318/images.zip)

yrabbit, 2023-12-20:
I continued my experiments - I made virtual Linux, took the current OSS and compiled not even Himbaechel, but legacy blinky.
[out.zip](https://github.com/YosysHQ/apicula/files/13721632/out.zip)

https://github.com/YosysHQ/apicula/assets/6075465/df974fc0-ae11-4db0-9f44-242b29d8ca3c

Next, I took my poor damaged tangnano9k, hoping that it wouldn’t burn anything for me

https://github.com/YosysHQ/apicula/assets/6075465/faea0501-9376-4430-827e-a2f6e622a798

ncboot, 2023-12-20:
Thank You for support. I will redo all my experiments.

marsohod4you, 2023-12-20:
Thank You for Your efforts. I will redo my experiments locally.
I think I miss something. But I do not know what yet.
Besides that, I must admit, Your images for blinky and shifter work on my tangnano9k properly.
PS: Do You know any tool which can be used to view json files out of Yosys as a schema? 
And also same graph viewer for json from nextpnr.
I was trying to use nexpnr with flag "--gui" but it crashes for me.

yrabbit, 2023-12-20:
You can experiment with this solution:
https://github.com/YosysHQ/nextpnr/issues/1255

Sorry, but there are not enough hands to keep the GUI up to date, especially since it is focused on legacy, and for Himbaechel a lot will have to be changed.

But you can send your json (both synth and pnr) here - I'll take a look.

yrabbit, 2025-03-25:
I don't think the legacy nextpnr is relevant anymore, so I think it's worth closing. As usual you are free to reopen if the problem is worth it.

closed 2025-03-25 by yrabbit