YosysHQ/apicula #216: szfpga board support

tpimh, 2023-12-21:
@yrabbit added support for a szfpga board in the commit d350cb7cee68144fdef159e8104bb7652ed91fdc. However, it is not mentioned anywhere in the documentation. Should the documentation be updated to add mention of this specific board?

The board has GW1NR-LV9LQ144PC6/I5, so I suppose it's [this product](https://www.szfpga.com/mod_product-view-p_id-114.html).

yrabbit, 2023-12-21:
Well, yes, this is the board. But I just don’t think that anyone except me has it, and I added it so that I wouldn’t have to apply a patch with its support every time.
[image]

tpimh, 2023-12-27:
Would it be okay to just add a mention of this board support to readme?

PR #218 (2023-12-27): Doc. Mention SZFPGA (GW1NR-9) board as supported.
Closes #216
closed 2023-12-27 by yrabbit