YosysHQ/apicula #217: Please use a crc library that provides a pure Python implementation

whitequark, 2023-12-22:
Right now, Apycula cannot be used with [Pyodide](https://pyodide.org/en/stable/):

```
>>> await micropip.install('Apycula==0.10.0')
Traceback (most recent call last):
  File "<console>", line 1, in <module>
  File "/lib/python3.11/site-packages/micropip/_commands/install.py", line 142, in install
    await transaction.gather_requirements(requirements)
  File "/lib/python3.11/site-packages/micropip/transaction.py", line 204, in gather_requirements
    await asyncio.gather(*requirement_promises)
  File "/lib/python3.11/site-packages/micropip/transaction.py", line 211, in add_requirement
    return await self.add_requirement_inner(Requirement(req))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lib/python3.11/site-packages/micropip/transaction.py", line 300, in add_requirement_inner
    await self._add_requirement_from_package_index(req)
  File "/lib/python3.11/site-packages/micropip/transaction.py", line 347, in _add_requirement_from_package_index
    await self.add_wheel(wheel, req.extras, specifier=str(req.specifier))
  File "/lib/python3.11/site-packages/micropip/transaction.py", line 385, in add_wheel
    await self.gather_requirements(wheel.requires(extras))
  File "/lib/python3.11/site-packages/micropip/transaction.py", line 204, in gather_requirements
    await asyncio.gather(*requirement_promises)
  File "/lib/python3.11/site-packages/micropip/transaction.py", line 208, in add_requirement
    return await self.add_requirement_inner(req)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lib/python3.11/site-packages/micropip/transaction.py", line 300, in add_requirement_inner
    await self._add_requirement_from_package_index(req)
  File "/lib/python3.11/site-packages/micropip/transaction.py", line 339, in _add_requirement_from_package_index
    wheel = find_wheel(metadata, req)
            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/lib/python3.11/site-packages/micropip/transaction.py", line 431, in find_wheel
    raise ValueError(
ValueError: Can't find a pure Python 3 wheel for 'crcmod'.
See: https://pyodide.org/en/stable/usage/faq.html#why-can-t-micropip-find-a-pure-python-wheel-for-a-package
You can use `await micropip.install(..., keep_going=True)` to get a list of all packages with missing wheels.
```

This is the only problematic dependency AFAICT (numpy is supported).

pepijndevos, 2023-12-22:
Any suggestions for such a library?

whitequark, 2023-12-23:
https://pypi.org/project/crc/

pepijndevos, 2023-12-28:
It's only used here so the change should be simple.

https://github.com/YosysHQ/apicula/blob/fedab4cf9871184c41e10ee6cdc6b306e3eb6a21/apycula/bslib.py#L5

I'm guessing we'll need these parameters, that's the main tricky part

whitequark, 2023-12-29:
```python
>>> import crcmod.predefined
>>> crc16 = crcmod.predefined.mkPredefinedCrcFun('crc-16')
>>> crc16(b"abcd")
14743
>>> calc = crc.Calculator(crc.Configuration(width=16, polynomial=0x8005, reverse_input=True, reverse_output=True))
>>> calc.checksum(b"abcd")
14743
```

PR #223 (2024-01-20): Switch crcmod to crc
Fixed #217 and hopefully makes peoply trying to to bundle/install Apicual in weird ways a little happier for having to compile less C.
commit 02520a0f (2024-01-20): Switch crcmod to crc (#223)

* Switch crcmod to crc

Fixed #217 and hopefully makes peoply trying to to bundle/install
Apicual in weird ways a little happier for having to compile less C.

* the stupid thing isn't thread safe

* python scoping rules are dumb
closed 2024-01-20 by pepijndevos