YosysHQ/apicula #220: Better documentation for Sipeed nano 20k support

siraisisatoru, 2024-01-04:
Hi,

I recently got a Sipeed Tang nano 20k device to verify my coursework design in the Verilog system. Before getting this device, I used to learn Xilinx devices and the corresponding software eco-system.

I noticed some tutorials for Tang Nano 9K using OSS CAD Suite. I want to use it for my device but it constantly complains that the device is invalid.

The script I used:
```sh
yosys -D LEDS_NR=8 -p "read_verilog top.v; synth_gowin -json blinky.json"
DEVICE='GW2AR-LV18QN88C8/I7'  # change to your device
BOARD='tangnano20k' # change to your board
nextpnr-gowin --json blinky.json \
              --write pnrblinky.json \
              --device $DEVICE \
              --cst $BOARD.cst\
```

[image]

Then I recognised `nextpnr-gowin` is built on top of Apicula and thus I constantly monitored this repo and finally someone added Sipeed Tang nano 20k on the readme. This sounds to me like I should be able to run `nextpnr-gowin` for my device but it is not and still getting the same error. By digging into the issues, I found https://github.com/YosysHQ/apicula/issues/208 and they mentioned "nextpnr-himbaechel" while this is not being mentioned anywhere (at least not on Google and the readme file for https://github.com/YosysHQ/nextpnr). 

The question I am having now is, how can I build my code for Sipeed Tang nano 20k no matter using nextpnr-himbaechel or nextpnr-gowin? I would also want to know how to build the examples. A step-by-step tutorial like https://github.com/YosysHQ/apicula#getting-started for Sipeed Tang nano 20k would be appreciated.

Thank you all for the hard work.

yrabbit, 2024-01-04:
hi,
For starters, you can't use nextpnr-gowin to create images for Tangnano20k.
Further, you cannot use the current release of nextpnr to use himbaechel gowin - because it is a May 2023 release and there is nothing related to gowin there.

So you need to compile and install nextpnr from the master branch.

To do this, you take this master branch:
`git clone https://github.com/YosysHQ/nextpnr.git`

install all the necessary things ([nextpnr prerequisites](https://github.com/YosysHQ/nextpnr?tab=readme-ov-file#prerequisites))

```
cd nextpnr 
cmake . -DBUILD_PYTHON=OFF -DARCH="gowin;himbaechel" -DHIMBAECHEL_GOWIN_DEVICES=all
make 
sudo make install

```

after that you can go to the apicula examples directory and build examples:
```
cd XXXX-apicula-root/examples/himbaechel
make -f Makefile.himbaechel
```
At each stage there may be ambiguities or my mistakes, so ask questions if something does not work.

pepijndevos, 2024-01-04:
We should really add this to the readme, either here or in nextpnr or both, and maybe bug @gatecat for a release

yrabbit, 2024-01-04:
I am for the release with both hands because people come after OSS CAD and we send them to learn git, download a bunch of dependencies, etc.

P.S. Besides, OSS doesn’t support BSD anyway, and I’m a lousy advisor to them here.

siraisisatoru, 2024-01-06:
Hi @yrabbit. I appreciated your detailed descriptions. I am here to bring you the good news that I can build the examples under the `examples/himbaechel` directory and successfully upload the blinky script to my FPGA. Although there were some problems with installing on my Macbook (possibly multiple duplicated installs previously and somewhat confused my machine and myself), I finally installed all the requirements and libs in a docker container under Ubuntu and that works. I remember there were some missing libs when installing `nextpnr`. I may tidy up a list of libs that are required later here.
[image]

Again, I appreciate all your help.

yrabbit, 2024-01-06:
Glad to hear that.
I want to ask why you write blinky into flash?
Writing to memory is faster and saves rewrite cycles. Although there may be millions of them out there, I haven’t looked.

siraisisatoru, 2024-01-06:
Hi @yrabbit. I don't have a particular reason for that. This is my first time using this device and just want to verify it is functional. I also took https://wiki.sipeed.com/hardware/en/tang/Tang-Nano-Doc/flash-in-linux.html as a reference which wrote the binary into flash.

closed 2024-01-06 by siraisisatoru

ReubenkThomas, 2024-07-30:
I tried running `make -f Makefile.himbaechel`, which produced the following error at the end. Any suggestions on this? I'm also running this on MacOS

```
2.32. Executing JSON backend.

End of script. Logfile hash: aa017d52bd, CPU: user 0.34s system 0.02s
Yosys 0.43+22 (git sha1 118b2829d, c++ 14.0.3 -fPIC -Os)
Time spent: 30% 20x read_verilog (0 sec), 10% 4x proc_dff (0 sec), ...
nextpnr-himbaechel --json blinky-tangnano20k-synth.json --write blinky-tangnano20k.json --device GW2AR-LV18QN88C8/I7 --vopt family=GW2A-18C --vopt cst=tangnano20k.cst
dyld[29903]: Library not loaded: @rpath/QtOpenGL.framework/Versions/5/QtOpenGL
  Referenced from: <0E202A22-1C50-3D5A-9BE2-4DB14C46E944> /usr/local/bin/nextpnr-himbaechel
  Reason: tried: '/Library/Frameworks/QtOpenGL.framework/Versions/5/QtOpenGL' (no such file), '/System/Library/Frameworks/QtOpenGL.framework/Versions/5/QtOpenGL' (no such file, not in dyld cache)
make: *** [blinky-tangnano20k.json] Abort trap: 6
rm blinky-tangnano20k-synth.json
```

ReubenkThomas, 2024-09-05:
Running `make -f Makefile.himbaechel` causes this error:

```
2.31. Executing CHECK pass (checking for obvious problems).
Checking module top...
Found and reported 0 problems.

2.32. Executing JSON backend.

End of script. Logfile hash: a5a337374c, CPU: user 0.44s system 0.03s
Yosys 0.43+22 (git sha1 118b2829d, c++ 14.0.3 -fPIC -Os)
Time spent: 20% 20x read_verilog (0 sec), 12% 8x techmap (0 sec), ...
nextpnr-himbaechel --json blinky-clkdiv-tangnano20k-synth.json --write blinky-clkdiv-tangnano20k.json --device GW2AR-LV18QN88C8/I7 --vopt family=GW2A-18C --vopt cst=tangnano20k.cst
Info: Using uarch 'gowin' for device 'GW2AR-LV18QN88C8/I7'

Info: Reading constraints...
Info: Cell clk_i not found
Info: Net clk_i not found
Info: Cell rst_i not found
Info: Cell rst_i not found
Info: Cell TXD not found
Info: Cell TXD not found
Info: Cell RXD not found
Info: Cell RXD not found
Info: Cell LED_R not found
Info: Cell LED_G not found
Info: Cell LED_B not found
Info: Cell tlvds_p not found
Info: Cell tlvds_n not found
Info: Cell elvds_p not found
Info: Cell elvds_n not found
Info: Cell led[6] not found
Info: Cell led[7] not found
Info: Cell oser_out not found
Info: Cell pclk_o not found
Info: Cell fclk_o not found
Info: Cell fclk_i not found
Info: Cell data_i not found
Info: Cell q_o[0] not found
Info: Cell q_o[1] not found
Info: Cell q_o[2] not found
Info: Cell q_o[3] not found
Info: Cell q_o[4] not found
Info: Cell q_o[5] not found
Info: Cell q_o[6] not found
Info: Cell q_o[7] not found
Info: Cell LCD_CLK not found
Info: Cell LCD_HYNC not found
Info: Cell LCD_SYNC not found
Info: Cell LCD_DEN not found
Info: Cell LCD_R[0] not found
Info: Cell LCD_R[1] not found
Info: Cell LCD_R[2] not found
Info: Cell LCD_R[3] not found
Info: Cell LCD_R[4] not found
Info: Cell LCD_G[0] not found
Info: Cell LCD_G[1] not found
Info: Cell LCD_G[2] not found
Info: Cell LCD_G[3] not found
Info: Cell LCD_G[4] not found
Info: Cell LCD_G[5] not found
Info: Cell LCD_B[0] not found
Info: Cell LCD_B[1] not found
Info: Cell LCD_B[2] not found
Info: Cell LCD_B[3] not found
Info: Cell LCD_B[4] not found
Info: Cell LCD_CLK not found
Info: Cell LCD_HYNC not found
Info: Cell LCD_SYNC not found
Info: Cell LCD_DEN not found
Info: Cell LCD_R[0] not found
Info: Cell LCD_R[1] not found
Info: Cell LCD_R[2] not found
Info: Cell LCD_R[3] not found
Info: Cell LCD_R[4] not found
Info: Cell LCD_G[0] not found
Info: Cell LCD_G[1] not found
Info: Cell LCD_G[2] not found
Info: Cell LCD_G[3] not found
Info: Cell LCD_G[4] not found
Info: Cell LCD_G[5] not found
Info: Cell LCD_B[0] not found
Info: Cell LCD_B[1] not found
Info: Cell LCD_B[2] not found
Info: Cell LCD_B[3] not found
Info: Cell LCD_B[4] not found
Info: Cell tmds_clk_p not found
Info: Cell tmds_clk_p not found
Info: Cell tmds_clk_n not found
Info: Cell tmds_clk_n not found
Info: Cell tmds_d_p[0] not found
Info: Cell tmds_d_p[0] not found
Info: Cell tmds_d_n[0] not found
Info: Cell tmds_d_n[0] not found
Info: Cell tmds_d_p[1] not found
Info: Cell tmds_d_p[1] not found
Info: Cell tmds_d_n[1] not found
Info: Cell tmds_d_n[1] not found
Info: Cell tmds_d_p[2] not found
Info: Cell tmds_d_p[2] not found
Info: Cell tmds_d_n[2] not found
Info: Cell tmds_d_n[2] not found
Info: Cell div_led not found
Info: Create constant nets...
Info: Modify LUTs...
Info: Pack IOBs...
Info: Pack diff IOBs...
Info: Pack IO logic...
Info: Pack DESER16 logic...
Info: Pack GSR...
Info: Pack wide LUTs...
Info: Packed MUX2_LUT8:0, MUX2_LU7:0, MUX2_LUT6:0, MUX2_LUT5:0
Info: Pack ALUs...
Info: Pack PLL...
Info: Pack RAMs...
Info: Pack BSRAMs...
Info: Pack DSP...
Info: Pack INV...
Info: Pack buffered nets...
Info: Checksum: 0x28b2e97b

Info: Device utilisation:
Info:                    VCC:       1/      1   100%
Info:                    IOB:       8/    384     2%
Info:                   LUT4:      11/  20736     0%
Info:               IOLOGICI:       0/    384     0%
Info:               IOLOGICO:       0/    384     0%
Info:              MUX2_LUT5:       0/  10368     0%
Info:              MUX2_LUT6:       0/   5184     0%
Info:              MUX2_LUT7:       0/   2592     0%
Info:              MUX2_LUT8:       0/   2592     0%
Info:                    ALU:     166/  15552     1%
Info:                    GND:       1/      1   100%
Info:                    DFF:     153/  15552     0%
Info:              RAM16SDP4:       0/    648     0%
Info:                  BSRAM:       0/     46     0%
Info:                 ALU54D:       0/     24     0%
Info:        MULTADDALU18X18:       0/     24     0%
Info:           MULTALU18X18:       0/     24     0%
Info:           MULTALU36X18:       0/     24     0%
Info:              MULT36X36:       0/     12     0%
Info:              MULT18X18:       0/     48     0%
Info:                MULT9X9:       0/     96     0%
Info:                 PADD18:       0/     48     0%
Info:                  PADD9:       0/     96     0%
Info:                    GSR:       1/      1   100%
Info:                    OSC:       0/      1     0%
Info:                   rPLL:       0/      2     0%
Info:                   BUFG:       1/     24     4%

Info: Placed 0 cells based on constraints.
ERROR: Unable to place cell 'hcount[4].my_div2', no BELs remaining to implement cell type 'CLKDIV2'
0 warnings, 1 error
make: *** [blinky-clkdiv-tangnano20k.json] Error 255
rm blinky-osc-tangnano20k.json blinky-osc-tangnano20k-synth.json blinky-oddr-tangnano20k-synth.json blinky-oddr-tangnano20k.json tlvds-tangnano20k-synth.json blinky-tbuf-tangnano20k.json elvds-tangnano20k.json blinky-tangnano20k.json shift-tangnano20k.json tlvds-tangnano20k.json blinky-tangnano20k-synth.json oddr-tlvds-tangnano20k-synth.json shift-tangnano20k-synth.json elvds-tangnano20k-synth.json oddr-tlvds-tangnano20k.json blinky-tbuf-tangnano20k-synth.json blinky-clkdiv-tangnano20k-synth.json
```

Any suggestions on how to fix?

Seyviour, 2024-09-05:
Could you try building himbaechel from source? The current release (and perhaps the version you have) don't have support for CLKDIV/CLKDIV2.

yrabbit, 2024-09-05:
And if you absolutely need CLKDIVs, nextpnr master has a bug related to creating clock PIPs.

So you'll either have to wait for the fix to be merged 
https://github.com/YosysHQ/nextpnr/pull/1358
or fix one line yourself.