YosysHQ/apicula #225: gowin_pack hangs on OSER10 - ELVDS_OBUF combo on Tang Nano 9K

regymm, 2024-01-21:
Hi, I'm recently trying to have HDMI output on Gowin Tang Nano 9K using project apicula, but I encountered the problem of gowin_pack hangs and falling into PDB, saying something like this: 

```
Info: [ 82567,  82568) | 
Info: [ 82568,  82569) | 

8 warnings, 0 errors

Info: Program finished normally.
SPINE5 SPINE13 not found in tile 1 47
> /oss-cad-suite/lib/python3.11/site-packages/Apycula-0.10.1.dev5+g153fc97-py3.11.egg/apycula/gowin_pack.py(1236)route()
-> continue
(Pdb)
```

I wonder what the problem might be? 

The verilog code, cst and makefile are at https://gist.github.com/regymm/93aa01486ad10a185976f2e665151e66
Note that in verilog code line 43/44, I've tried multiple clock configurations, including clkpix/clk_250, clkpix/clk_TMDS, but all have the same problem. 

Btw, I'm kinda curious if there's any existing HDMI demo using the open-source toolchain? I should be able to use 25M/250M and manually write a TMDS decoder as in the fpga4fun example, but it won't pass timing (in Gowin IDE, the Fmax is around ~180M and HDMI won't display).

yrabbit, 2024-01-21:
I don’t think that we will be able to surpass Gowin IDE - it is like a standard with which we compare our developments.

But to the point: OSER10 ELVDS works in the new router architecture - Himbaechel, in just two hours I will add the example oser10-elvds-tangnano9k to the standard apicula distribution.

https://github.com/YosysHQ/apicula/assets/6075465/cd6e633e-8bcb-4ca4-addc-5afcd78c75a9

Try your design with the apicula master branch and the nextpnr master branch (you can see how to install it in https://github.com/YosysHQ/apicula/issues/220#issuecomment-1876682997).
If your design still can’t compile, then let me know and we’ll figure it out.

regymm, 2024-01-21:
Oh I didn't mean apicula's performance, in Gowin IDE my 250 MHz encoder logic can only run at 180 MHz, which is not even close. So maybe Gowin just expects us to use OSER10. 

Can I ask how to install apicula master branch? The makefile seems to require the Gowin IDE and fuzzing. Do I need to do this (instead of using a pre-built library, etc)? 

I tried using nextpnr master branch with himbaechel, and apicula of version `pip install apycula`, and the design still has the same problem.

yrabbit, 2024-01-21:
If you already have a master nextpnr, then you will most likely be able to assemble your own design.

Try calling the desired router:
``` makefile
nextpnr-himbaechel --json $< --write $@ --device GW1NR-LV9QN88PC6/I5 --vopt family=GW1N-9C --vopt cst=$(filter-out $<,$^)
```

And in the .CST file, specify each pin from the differential pair separately:
```
IO_LOC "tmds_d_p" 71;
IO_PORT "tmds_d_p" PULL_MODE=NONE DRIVE=8;
IO_LOC "tmds_d_n" 70;
IO_PORT "tmds_d_n" PULL_MODE=NONE DRIVE=8;
```

(yes, I know about Gowin syntax, but I’m very bad at writing parsers - I’m much worse at it than researching the insides of a chip. Maybe I’ll improve it in the future.)

regymm, 2024-01-21:
Oh I should have used nextpnr-himbaechel. Now HDMI is working. Thanks for your quick help!

closed 2024-01-21 by regymm

pepijndevos, 2024-01-21:
Cool! If you'd like to contribute a working HDMI example it'd be much appreciated.

regymm, 2024-01-22:
Of course! The code is based on fpga4fun's example, at https://github.com/FPGAOL-CE/user-examples/tree/main/hdmi . I'm using this dockerized toolchain https://github.com/FPGAOL-CE/osstoolchain-docker-things/blob/master/gowin/Dockerfile.gowin-gocd . HDMI clocks are done by generate 25M and 250M clk from PLL, then divide the 250M by 2 using a register. That's the closest I can get to without CLKDIV. Since the 125M drives only the OSER10, this untimed part is not likely to fail.