YosysHQ/apicula #232: Failed to route net Gowin_rPLL_9Mhz.clkout with nextpnr-himbaechel

mitnk, 2024-01-28:
Hello,

I'm trying build the [LCD demo with tang-nano-9k](https://github.com/mitnk/nextpnr-9k-lcd), using [this command](https://github.com/mitnk/nextpnr-9k-lcd/blob/main/build-pnr.sh), and get following error:
```
Info: No Fmax available; no interior timing paths found in design.
Info: Checksum: 0x911b5b1c
Info: Routing globals...
Warning: Failed to route net 'Gowin_rPLL_9Mhz.clkout' from X25Y9/TRPLL0CLK0 to X33Y8/A6 using dedicated routing.
Info:     routed net 'Gowin_rPLL_9Mhz.clkin' using global resources

Info: Routing..
Info: Setting up routing queue.
ERROR: Assert `net_info->wires.count(wire)' failed in /Users/mitnk/fpga/nextpnr/common/route/router1.cc:345.
ERROR: Routing design failed.
```

Could anyone point me what's the next step I should do?

yrabbit, 2024-01-28:
In a few minutes I will make corrections in nextpnr and the networks will be routed correctly.
In the meantime, one question: are you sure that your design will show something on the screen? I ask because I have a white screen and if you are confident in the design, then the problems with the networks are much deeper than I thought.

mitnk, 2024-01-29:
Yes, it should look like this one:

[image]

[link](https://github.com/sipeed/TangNano-9K-example#lcd_43)

yrabbit, 2024-01-29:
hmm... this is their picture, but have you tried compiling this project in the IDE?
Or just download their ready-made .fs and flash it in Tangnano9k?

https://github.com/sipeed/TangNano-9K-example/tree/main/lcd_4.3

[image]

Perhaps my board just died completely.

yrabbit, 2024-01-29:
If you have the apicula repository installed, then you can now assemble the example bsram-SDPB-tangnano9k.fs which is more beautiful and everything is fine with VGA and wires :wink: 

```shell
cd wherever-you-installed-the-apicula-repository/apicula/examples/himbaechel
make -f Makefile.himbaechel bsram-SDPB-tangnano9k.fs
```

mitnk, 2024-01-29:
> make -f Makefile.himbaechel bsram-SDPB-tangnano9k.fs

This works the the two little spirits show up on my LCD. (with a Mac with M2 chip)

> have you tried compiling this project in the IDE?

Yes, I did compiled the source with Gowin IDE on Windows and it works (never used their fs file).

yrabbit, 2024-01-29:
If you don’t want to compile nextpnr yourself, then remove references to all unused PLL outputs of the .CLKOUTP type and your project will compile right now.
It's strange that in my case everything is white...

mitnk, 2024-01-29:
Thanks @yrabbit for the help!

I'll try compile later today using the branch you drafted.

Had a quick skim on command in `Makefile.himbaechel`, it involves some `vh` files in, but there seems not corresponding think in [my pll](https://github.com/mitnk/nextpnr-9k-lcd/tree/main/gowin_rpll), not sure if it is because of the Gowin IP thing does not "linked" or not.

yrabbit, 2024-01-29:
Well, apicula is designed for many boards with different quartz and therefore the PLL parameters are placed in separate files.

yrabbit, 2024-01-29:
I looked at the generated signals:
Sorry for not changing the label, but the connection is this:

channel 0 - pin 35 LCD_LCK
channel 1 - pin 34 LCD_SYNC
channel 2 - pin 33 LCD_DEN
channel 3 - pin 40 LCD_HYNC

The apicula:  LCD_CLK 9MHz, a very beautiful and uniform vertical sync signal ~ 58 frames per second.

https://github.com/YosysHQ/apicula/assets/6075465/88092f31-99f0-4076-95d9-a74e86f44ebd

Gowin option - LCD_CLK is also 9MHz, but instead of a vertical sync signal there is terror that flaps in the night.

https://github.com/YosysHQ/apicula/assets/6075465/4f64584a-c00b-4d36-882b-0fdc26a799b1

[image]

My display just can’t handle such tricks, so I think there are no errors in software just different displays - your display can work with such vertical sync, but mine can't. You can close the issue. :wink:

mitnk, 2024-01-29:
I tried your nextpnr branch, it works! Thanks for your time @yrabbit 

I noticed there are some noise in the LCD with apicula, compared to Gowin compiled. But the fs apicula produced is good enough.

closed 2024-01-29 by mitnk

yrabbit, 2024-01-29:
Oh, this is quite expected - in your first message it is indicated that the router cannot route from the PLL output to X33Y8/A6. Here the destination is not a clock input or IO pin and such networks are currently given to the general purpose router. Perhaps in the future I will complicate the dedicated router or not - I somehow don’t really like using the clock as an input for the LUT :smile: