YosysHQ/apicula #243: Cant find how to compile for tang nano 20k

H4-b, 2024-04-27:
The tang nano 20k  board is marked as supported by the project, by there is no example like for other boards and trying to use it like the primer 20k doesn't work :

[image]

i cant seem to find how to get the version number of apicula, but it's from the latest release of oss-cad-suite (2024-04-27)

What am i doing wrong? Is it just an outdated version?

yrabbit, 2024-04-27:
The apicula kit is full of examples:
https://github.com/YosysHQ/apicula/blob/master/examples/himbaechel/Makefile.himbaechel

only they use nextpner-himbaechel, which you can probably install from the master branch. Or you can wait until it works as part of oss-cad-suite 
https://github.com/YosysHQ/nextpnr/pull/1315#pullrequestreview-2025164708

H4-b, 2024-04-27:
Hello!
Thank you for your quick response.
I found that the oss-cad-suite from  [lushaylabs ](https://github.com/lushaylabs/oss-cad-suite-build) has nextpnr-himbaechel (and a [pull request](https://github.com/YosysHQ/oss-cad-suite-build/pull/109) to the main branch). I just tested it and it seems to work just fine for now. 
Thank you for the examples, i didn't think to dive in the folders of the examples.

closed 2024-04-27 by H4-b

pepijndevos, 2024-04-27:
We should really delete the old target asap

yrabbit, 2024-05-01:
> We should really delete the old target asap
Well the process is started: https://github.com/YosysHQ/nextpnr/pull/1318

pepijndevos, 2024-05-01:
I don't think we really depend on nextpnr. Since the old target isn't getting any new features they can continue to build nextpnr-gowin with the latest supported release.

yrabbit, 2024-05-01:
Yes, our problems start when they build nextpnr-gowin only ;)