YosysHQ/apicula #244: Tang Nano 9k: rPLL is not working at all

tmssngr, 2024-04-29:
- clone the branch `rpll-test` from https://github.com/tmssngr/fpga
- with Gowin IDE open the `video-signal/video-signal.gprj`, run "Run All", upload to the Tang Nano 9k
-> the Tang Nano 9k counts the LEDs slowly up (and on the connected Video output a rectangle is shown)
- now edit `video-signal/src/build.cmd` to contain the correct path to oss-cad-suite (I'm tried with oss-cad-suite-240428)
- (with *zadig* ensure that for *JTAG Debugger (Interface 0)* replaced the driver to WinUSB)
- run `video-signal/src/build.cmd`
-> no LED counts up (no rect on the Video output shown)

Am I doing something obvious wrong or is the rPLL not supported (with this configuration)?

pepijndevos, 2024-04-29:
Could you try with #241 ?

tmssngr, 2024-04-29:
Which official build I should try (I can't build on my own)?

pepijndevos, 2024-04-29:
There isn't an official release with an unmerged pull request.
We do however compile a Python package on CI, as well as build the examples.
So if you can obtain a new enough nextpnr you could use it together with the provided python package.
If you are unable to obtain a nextpnr newer than 3 weeks, you can at least verify that the PLL examples in the himbaechel folder work correctly.

If you scroll down to artifacts you will find examples-main-master with ready to use bitstreams, and python-dist for use with a recent nextpnr.
https://github.com/YosysHQ/apicula/actions/runs/8593362463?pr=241

yrabbit, 2024-04-29:
`Still waiting for nextpnr-himbaechel to be added:
https://github.com/YosysHQ/oss-cad-suite-build/pull/109

And so the LEDs are blinking, I can take a video signal from the pins, but something tells me that everything is fine there too.

`` shell
#!/bin/sh
yosys -p "read_verilog top.v gowin_rpll/gowin_rpll.v VbsGenerator.v; synth_gowin -top top -json synth.json"
nextpnr-himbaechel --json synth.json --write project.json --device GW1NR-LV9QN88PC6/I5 --vopt family=GW1N-9C --vopt cst=tangnano9k.cst
gowin_pack -d GW1N-9C -o project.fs project.json
openFPGALoader -c ft2232 project.fs
```

https://github.com/YosysHQ/apicula/assets/6075465/c17cd8b6-c6e2-415b-8050-78d356e629b2

tmssngr, 2024-04-30:
That looks perfectly fine. I'll wait until an official build will contain the fix/change.

pepijndevos, 2024-07-28:
fixed in #241

closed 2024-07-28 by pepijndevos