YosysHQ/apicula #267: Are BRAMs supported?

stacksmith, 2024-07-30:
Having trouble inferring BRAMS on a tang nano9K...

yrabbit, 2024-07-30:
Do you have a broken example and a compilation log?

You can look into the directory https://github.com/YosysHQ/apicula/tree/master/examples/himbaechel - there is not only direct use of memory primitives (SP, SDP, DP.. etc) but also inferring as part of the RISCV emulator, which works fine on Tangnano9k - https://github.com/YosysHQ/apicula/blob/master/examples/himbaechel/femto-riscv-18.v

stacksmith, 2024-07-30:
Thank you, I was hoping to hear that it works.  I didn't have working code until today, just got a 65c02 core up with a 2K rom and a 2K RAM using the awful gowin toolchain (takes just over a minute to build!).  Tomorrow will rebuild with apicula.  Hoping for a faster turnaround, as a minute is way too slow.

Thanks again

stacksmith, 2024-07-30:
OK, seems to work fine.  Thank you.

closed 2024-07-30 by stacksmith