YosysHQ/apicula #278: trouble running the example files for tang sipeed nano 20k

sriram021, 2024-09-05:
```
nextpnr-himbaechel --json blinky.json \
                   --write pnrblinky.json \
                   --device $DEVICE \
                   --vopt family=GW2A-18 \
                   --vopt cst=$BOARD.cst
gowin_pack -d GW2A-18 -o pack.fs pnrblinky.json
Info: Using uarch 'gowin' for device 'GW2AR-LV18QN88C8/I7'
ERROR: No package for partnumber GW2AR-LV18QN88C8/I7
0 warnings, 1 error
Traceback (most recent call last):
  File "/Users/---/Desktop/oss-cad-suite/libexec/gowin_pack", line 33, in <module>
    sys.exit(load_entry_point('Apycula==0.13.dev10+gbf38174', 'console_scripts', 'gowin_pack')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/apycula/gowin_pack.py", line 2842, in main
    with open(args.netlist) as f:
         ^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'pnrblinky.json'
MacBook-Air examples % 
```
username redacted for privacy, having trouble running for tang sipeed nano 20k

yrabbit, 2024-09-05:
Specify the GW2A-18C not GW2A-18 family to run both nextpnr and gowin_pack.

sriram021, 2024-09-05:
```
nextpnr-himbaechel --json blinky.json \                                     
                   --write pnrblinky.json \
                   --device $DEVICE \
                   --vopt family=GW2A-18C \
                   --vopt cst=$BOARD.cst
gowin_pack -d GW2A-18C -o pack.fs pnrblinky.json

Info: Using uarch 'gowin' for device 'GW2AR-LV18QN88C8/I7'

Info: Reading constraints...
Info: Cell clk_i not found
Info: Net clk_i not found
Info: Cell key_i not found
Info: Cell key_i not found
Info: Cell rst_i not found
Info: Cell rst_i not found
Info: Cell TXD not found
Info: Cell TXD not found
Info: Cell RXD not found
Info: Cell RXD not found
Info: Cell LED_R not found
Info: Cell LED_G not found
Info: Cell LED_B not found
Info: Cell tlvds_p not found
Info: Cell tlvds_n not found
Info: Cell elvds_p not found
Info: Cell elvds_n not found
Info: Cell led[6] not found
Info: Cell led[7] not found
Info: Cell oser_out not found
Info: Cell pclk_o not found
Info: Cell fclk_o not found
Info: Cell fclk_i not found
Info: Cell data_i not found
Info: Cell q_o[0] not found
Info: Cell q_o[1] not found
Info: Cell q_o[2] not found
Info: Cell q_o[3] not found
Info: Cell q_o[4] not found
Info: Cell q_o[5] not found
Info: Cell q_o[6] not found
Info: Cell q_o[7] not found
Info: Cell LCD_CLK not found
Info: Cell LCD_HYNC not found
Info: Cell LCD_SYNC not found
Info: Cell LCD_DEN not found
Info: Cell LCD_R[0] not found
Info: Cell LCD_R[1] not found
Info: Cell LCD_R[2] not found
Info: Cell LCD_R[3] not found
Info: Cell LCD_R[4] not found
Info: Cell LCD_G[0] not found
Info: Cell LCD_G[1] not found
Info: Cell LCD_G[2] not found
Info: Cell LCD_G[3] not found
Info: Cell LCD_G[4] not found
Info: Cell LCD_G[5] not found
Info: Cell LCD_B[0] not found
Info: Cell LCD_B[1] not found
Info: Cell LCD_B[2] not found
Info: Cell LCD_B[3] not found
Info: Cell LCD_B[4] not found
Info: Cell LCD_CLK not found
Info: Cell LCD_HYNC not found
Info: Cell LCD_SYNC not found
Info: Cell LCD_DEN not found
Info: Cell LCD_R[0] not found
Info: Cell LCD_R[1] not found
Info: Cell LCD_R[2] not found
Info: Cell LCD_R[3] not found
Info: Cell LCD_R[4] not found
Info: Cell LCD_G[0] not found
Info: Cell LCD_G[1] not found
Info: Cell LCD_G[2] not found
Info: Cell LCD_G[3] not found
Info: Cell LCD_G[4] not found
Info: Cell LCD_G[5] not found
Info: Cell LCD_B[0] not found
Info: Cell LCD_B[1] not found
Info: Cell LCD_B[2] not found
Info: Cell LCD_B[3] not found
Info: Cell LCD_B[4] not found
Info: Cell tmds_clk_p not found
Info: Cell tmds_clk_p not found
Info: Cell tmds_clk_n not found
Info: Cell tmds_clk_n not found
Info: Cell tmds_d_p[0] not found
Info: Cell tmds_d_p[0] not found
Info: Cell tmds_d_n[0] not found
Info: Cell tmds_d_n[0] not found
Info: Cell tmds_d_p[1] not found
Info: Cell tmds_d_p[1] not found
Info: Cell tmds_d_n[1] not found
Info: Cell tmds_d_n[1] not found
Info: Cell tmds_d_p[2] not found
Info: Cell tmds_d_p[2] not found
Info: Cell tmds_d_n[2] not found
Info: Cell tmds_d_n[2] not found
Info: Cell div_led not found
Info: Create constant nets...
Info: Modify LUTs...
Info: Pack IOBs...
ERROR: Unconstrained IO:key_IBUF_I
0 warnings, 1 error
Traceback (most recent call last):
  File "/Users/---/Desktop/oss-cad-suite/libexec/gowin_pack", line 33, in <module>
    sys.exit(load_entry_point('Apycula==0.13.dev10+gbf38174', 'console_scripts', 'gowin_pack')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.11/site-packages/apycula/gowin_pack.py", line 2842, in main
    with open(args.netlist) as f:
         ^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'pnrblinky.json'
```
sorry for asking a lot, but here's another issue i'm facing after trying your suggestion

yrabbit, 2024-09-05:
All Input/Output/Inout in Top Module must be described in the .cst file, in this case, "key" is not described.
Are you saying an example from the supply of Apicula? Which one? Since this must be corrected immediately.

sriram021, 2024-09-05:
i'm using the blinky.v example

sriram021, 2024-09-05:
I have a doubt, so going forward, if i want to synthesize my own designs, i have to write my own cst files?

yrabbit, 2024-09-05:
Of course, in the end it's your decision what to use the board pins for and what to name them. 

Supplied in the blinky.v set does not contain "key" input:
https://github.com/YosysHQ/apicula/blob/master/examples/himbaechel/blinky.v

sriram021, 2024-09-05:
I kind of understand what to do, will get back to let you

sriram021, 2024-09-06:
got it working, thanks!

closed 2024-09-06 by sriram021